lmarzen/esp32-weather-epd #116: Refresh times for tri colour displays

Asdf1qaz, 2024-07-22:
Gday mate. Love the look of project so far.
Sorry for posting it here, not really sure where else to post.
Just looking at buying all the parts and trying to sauce a display in AU.
It seems that I can get the GDEY075Z08 for about the same price as the black and white one.
My questions are, first off, will it work with your code.
Secondly, will a refresh take the 26 seconds? Or just using black and white only take the 1.3 secs?
Saying that, having any alerts at the top written in red might be a cool feature.

lmarzen, 2024-07-23:
The GDEY075Z08 is supported, see the "panel support" section in the readme. https://github.com/lmarzen/esp32-weather-epd/tree/main?tab=readme-ov-file#hardware

I have not tested the GDEY075Z08 panel. Support has been added for it by the community. I'm not sure the exact refresh time. The black and white takes about 4 seconds.

Asdf1qaz, 2024-07-25:
Cheers mate.
Going with the black and white one!

closed 2024-07-25 by Asdf1qaz