lmarzen/esp32-weather-epd #117: Add dew point and moon phase

Asdf1qaz, 2024-08-01:
Hay mate,
Feature requests if I could.
I have had a look at the code, and im still starting out, so this is a bit above me. I will probably break it if I changed anything.

I think it would be beneficial to add the current moon phase (included in the OWM API response as 'daily.moon_phase' and have a value of "0 and 1 are 'new moon', 0.25 is 'first quarter moon', 0.5 is 'full moon' and 0.75 is 'last quarter moon'. "

Also, the Dew Point is beneficial for two reasons, one, its a way better way to state how the air moisture 'feels' and two, a way to tell if it will be dewy over night.
This is included as two values from the API as: 'current.dew_point' and 'hourly.dew_point'

IMO I think 'current dew point' and 'moon phase' could go under sun rise and sunset, if you slightly move the 'current temp' and 'weather icon' up a touch/smaller.

Looking at Pivoine31's code, I love how he has put the rain pop and amount under each day in
https://github.com/pivoine31/esp32-weather-epd-pvmod/blob/main/show.mod/pop_and-vol.jpg

So for the 'hourly dew point' if the pop and rain amount is implemented, the daily max and min could go between the temp and rain data in the 5 day forecast? There is a lot of space between that and the graph that it could move into.

Pivoine31 has also done some good work with setting a lower infill amount in the graph if there is too much POP predicted (which makes displays hate life), a WebUI to set things up and put a rain amount AND POP in the graph (as you  can see in that pic above). So maybe they could get implemented too?

lmarzen, 2024-08-01:
Those are all great suggestions. I will take a look at what you suggested. I have been working on my own WebUI in the background for some time now. I have been busy with school, work, and life lately, but when I find some free time, you expect to see some of these features.

Asdf1qaz, 2024-08-01:
Sounds good. Its a fantastic project you have here BTW.

Also, sorry but I made a mistake with the hourly.dew_point. It is only for up to 48 hours. So maybe that could go in the graph as a dotted line, like the temp line?

That would also remove the need to display the current dew point to the left. So, if the mood phase is a go, maybe put the 'current.clouds' % on the other side, to make is pretty and fit in with the rest?

I know I'm asking a lot here and not sure if it should be in a new issue, but to top it off, maybe also display the inside dew point, as read from the BME280.
So it looks pretty, have the bottom row on the left in a box, as they are all read locally, and have all three measurements in there?

Dew point can be calculated with:
DP = T°C – ((100 – RH)/5)

There is a good page on this at:
https://learnmetrics.com/dew-point-calculator-chart-formula/

lmarzen, 2026-01-05:
Moon phase has been added in https://github.com/lmarzen/esp32-weather-epd/pull/208, Dewpoint still needs to be added.