lmarzen/esp32-weather-epd #118: Code Setup and preparation

ltwin8, 2024-08-02:
Hi, I would like to build this project.

I was browsing the code and was wondering how to set 24h format for time and the gps coordinates or LATITUDE/LONGITUDE, didn’t find it in the .inc file for Germany.

thank you in advance.

great project 👍

lmarzen, 2024-08-02:
See config.cpp
https://github.com/lmarzen/esp32-weather-epd/blob/9d9cdf33771ecc9f4d864853065588fc0b24deb3/platformio/src/config.cpp#L79

ltwin8, 2024-08-02:
Thank you very much :)

could you tell me the where to find time format setting? I found time related stuff in the .inc, but I don’t get how to use it and the meaning of the letters like %T

lmarzen, 2024-08-02:
You should not change the time format in .inc. You change in in config.cpp. There is a comment with a link that explains what the % stuff means... https://man7.org/linux/man-pages/man3/strftime.3.html

The time stuff in .inc is standard for that locale and is used to generate the correct output when you use abbreviations like %D for example whoch expands to mm/dd/yyyy in the US, but dd/mm/yyyy in many other countries.

ltwin8, 2024-08-02:
Okay, thank you very much 👍