lmarzen/esp32-weather-epd #119: E-Paper hat rev2.3

ltwin8, 2024-08-03:
I have read a bit about the support of the driver hardware, the people reporting issues with the rev2.3 hardware could try to "re-update" their driver hardware. the inductor (old 68uH new 10uH) seems not to be the issue but the MOSFET Q1 seems to be at fault here. 

https://files.waveshare.com/upload/8/8e/E-Paper_Driver_HAT.pdf                               (2.3)
https://files.waveshare.com/upload/8/87/E-Paper-Driver-HAT-Schematic.pdf              (2.2)

The transistor Q1 is responsible for creating the driver voltages, those ones are buffered in a 4.7uF MLCC, during dithering this "reservoir" seems to run dry and the DCDC can not keep up "refilling" it

the setting of RSEL to 470mR should allow higher current (after all its the sense shunt) but the transistor Q1 has a way to high Rds_on (around 3R) compared to the old one (around 0.2R) so i recommend to solder a transistor of this type "SI1308EDL-T1-GE3" or simmilar.

important is the following:
Vds >= 30V
Vgs_th <= 2V
Rds_on <= 0.25R at 2.5V-3V Vgs
low total gate charge, less than 2.5nC

this transistor is likely the one build into the old board: 
https://www.mouser.de/ProductDetail/Vishay-Semiconductors/SI1308EDL-T1-GE3?qs=bX1%252BNvsK%2FBramh9tgpOaEw%3D%3D

if you search for alternaives filter the results at your supplier of choice with the above given values and there you go... to finish the decision open the datasheet, search pinout and compare to datasheet of SI1308EDL-T1-GE3 (available at the link above)

To those not very experienced with soldering or dont have rework equipment: the old MOSFETon the board can get destroyed, either remove its plastic with a nail clipper or simmilar, than heat the 3 pads simultaniously or put a blob of solder on each of the 3 pads and do round robin heating. to solder in the new one should be easy, tin the leads, put it with tweezers in place, shotrt heat to each pad should be sufficient for reflowing

to make the effort in reviving the "new and better (sure...)" driver hat (2.3) can have some advantage because it does after all provide a layer of additional ESD protection for the screen.

if helpful maybe add this guide to the readme, as i think this "hack" is not so difficult to perform

Asdf1qaz, 2024-08-06:
Good work on this mate but those links you posted for the transistor location on the board are not working. Would you be able to post a pic of them?

Also, for Imarzen, I found this "The new HAT 2.3 from Waveshare will consume zero current when PWR is not pulled high, AND all control and data lines are set floating, set INPUT. Call display.end(); after display.hibernate(); and before setting processor to deep sleep."

Not sure how much power the V2.3 hats take when idle normally (I will test when I get home), but this might be good to implement so you can power it down just like how the BME280 is done

jordiuoc, 2024-08-06:
Hi!

Thank you for this project. It's super!

I wanted to share a potential improvement for e-paper hat rev.2.3. If you change the function `_InitDisplay` in the file `GxEPD2_750_T7.cpp` to:

```cpp
void GxEPD2_750_T7::_InitDisplay() 
{
  if (_hibernating) _reset();

  // POWER SETTING
  _writeCommand(0x01);
  _writeData(0x07);    // VDS_EN, VDG_EN
  _writeData(0x07);    // VGH=20V, VGL=-20V
  _writeData(0x3E);    // VDH=14V (experiment with 0x3E, 0x40)
  _writeData(0x3E);    // VDL=-14V (experiment with 0x3E, 0x40)
  
  // PANEL SETTING
  _writeCommand(0x00);
  _writeData(0x1F);    // KW: 3F, KWR: 2F, BWROTP: 0F, BWOTP: 1F
  
  // TRES SETTING
  _writeCommand(0x61);
  _writeData(WIDTH / 256);  // source 800
  _writeData(WIDTH % 256);
  _writeData(HEIGHT / 256); // gate 480
  _writeData(HEIGHT % 256);
  
  _writeCommand(0x15);
  _writeData(0x00);
  
  // VCOM AND DATA INTERVAL SETTING
  _writeCommand(0x50);
  _writeData(0x27);    // LUTKW, N2OCP: copy new to old (experiment with 0x27, 0x2A, 0x2B)
  _writeData(0x06);    // Experiment with 0x06, 0x08
  
  // TCON SETTING
  _writeCommand(0x60);
  _writeData(0x22);
}
```

I found that experimenting with the VCOM and Data Interval settings was necessary. This file is part of a library, so modifying it might be tricky.

Additionally, I am encountering an issue with refreshing. I can still see old data after refreshing with new data. This might be a problem in the `initDisplay` function in the `renderer.cpp` file. If I find a solution, I will let you know. Meanwhile, any advice is welcome.

Best regards,

ltwin8, 2024-08-09:
can you confirm/meassure if voltages stay in check during updating? eink is very sensitive to this because it relies on stati charge to move particels precisely

ZinggJM, 2024-08-11:
There is a modified version of the driver class for GDEY075T7 available on my branch work_in_progress:
https://github.com/ZinggJM/GxEPD2/blob/work_in_progress/src/gdey/GxEPD2_750_GDEY075T7.cpp

This version adds the Power Save (PWS) controller command to the initialization (line 342).
It solves an issue I have seen with dithered bitmaps.
Thank you for this report.

ltwin8, 2024-08-18:
> Hi!
> 
> Thank you for this project. It's super!
> 
> I wanted to share a potential improvement for e-paper hat rev.2.3. If you change the function `_InitDisplay` in the file `GxEPD2_750_T7.cpp` to:
> 
> ```c++
> void GxEPD2_750_T7::_InitDisplay() 
> {
>   if (_hibernating) _reset();
> 
>   // POWER SETTING
>   _writeCommand(0x01);
>   _writeData(0x07);    // VDS_EN, VDG_EN
>   _writeData(0x07);    // VGH=20V, VGL=-20V
>   _writeData(0x3E);    // VDH=14V (experiment with 0x3E, 0x40)
>   _writeData(0x3E);    // VDL=-14V (experiment with 0x3E, 0x40)
>   
>   // PANEL SETTING
>   _writeCommand(0x00);
>   _writeData(0x1F);    // KW: 3F, KWR: 2F, BWROTP: 0F, BWOTP: 1F
>   
>   // TRES SETTING
>   _writeCommand(0x61);
>   _writeData(WIDTH / 256);  // source 800
>   _writeData(WIDTH % 256);
>   _writeData(HEIGHT / 256); // gate 480
>   _writeData(HEIGHT % 256);
>   
>   _writeCommand(0x15);
>   _writeData(0x00);
>   
>   // VCOM AND DATA INTERVAL SETTING
>   _writeCommand(0x50);
>   _writeData(0x27);    // LUTKW, N2OCP: copy new to old (experiment with 0x27, 0x2A, 0x2B)
>   _writeData(0x06);    // Experiment with 0x06, 0x08
>   
>   // TCON SETTING
>   _writeCommand(0x60);
>   _writeData(0x22);
> }
> ```
> 
> I found that experimenting with the VCOM and Data Interval settings was necessary. This file is part of a library, so modifying it might be tricky.
> 
> Additionally, I am encountering an issue with refreshing. I can still see old data after refreshing with new data. This might be a problem in the `initDisplay` function in the `renderer.cpp` file. If I find a solution, I will let you know. Meanwhile, any advice is welcome.
> 
> Best regards,

why would you change this part of the library? the issue with rev2.3 hat hardware is the use of a bad spec'd mosfet

jordiuoc, 2024-08-19:
Hi!

You are absolutely right. I shouldn't touch the library. For now, I'm looking for the previous hardware version, although I haven't been able to find it in my area. Interisting video of applied science: https://www.youtube.com/watch?v=MsbiO8EAsGw

Just to let you know, I'm using this project to determine whether I should irrigate or not. Based on the humidity percentages of my field and the weather conditions, I decide if irrigation is needed. The use of these e-Ink screens has been extremely interesting to me, so I took advantage of your project. People have interesting hobbies, don't they? :-)

lmarzen, 2024-08-25:
Bumped zinggjm/GxEPD2 version 1.5.6 ->1.5.8 (or higher) which includes the mentioned fixes. (2567265d69843ad01c24204927f7dc18bc09e22d)

SyahfiraTasya, 2024-12-13:
i have tried with schematic

lmarzen, 2024-12-13:
@SyahfiraTasya Did that work for you?

lmarzen, 2025-01-06:
Closing since no response. Assuming zinggjm/GxEPD2 version 1.5.8 resolved the issue. Will reopen if otherwise indicated.

closed 2025-01-06 by lmarzen