lmarzen/esp32-weather-epd #120: Compiling failed in PlatformIO...

schroepfer, 2024-08-04:
Hello,
after compiling your code in VS-Code and PlatfotmIO i get the error "firmware.map: No such file or directory"

Please can you help me here?

Regards
Juergen

`Building in release mode
Linking .pio\build\dfrobot_firebeetle2_esp32e\firmware.elf
c:/.platformio/packages/toolchain-xtensa-esp32/bin/../lib/gcc/xtensa-esp32-elf/8.4.0/../../../../xtensa-esp32-elf/bin/ld.exe: cannot open map file C:/Users/Jürgen/Documents/PlatformIO/Projects/esp32-weather-epd-main/platformio/.pio/build/dfrobot_firebeetle2_esp32e/firmware.map: No such file or directory
collect2.exe: error: ld returned 1 exit status
*** [.pio\build\dfrobot_firebeetle2_esp32e\firmware.elf] Error 1
========================================================== [FAILED] Took 36.79 seconds ==========================================================

Environment                 Status    Duration
--------------------------  --------  ------------
dfrobot_firebeetle2_esp32e  FAILED    00:00:36.787
===================================================== 1 failed, 0 succeeded in 00:00:36.787 =====================================================`

lmarzen, 2024-08-04:
seems to be a bug with platformio...
https://community.platformio.org/t/firmware-map-no-sunch-file-or-directory/32302

schroepfer, 2024-08-05:
Thank you!
The problem ist my username with german umlaut: "open map file C:/Users/Jürgen/Documents/Plat...."
Moving the project to another place... it compiles!

closed 2024-08-05 by schroepfer