lmarzen/esp32-weather-epd #121: Battery Voltage reading error

Asdf1qaz, 2024-08-05:
Hay mate. Got it all working and its great.

Just found something though, the battery voltage reading is off.
At the moment, my battery (when idle, between screen refreshes) is sitting at 4.151 volts.
During screen refreshes, it goes down to 4.137-4.14 volts.

On the screen it displays 4.28 volts. (not sure when that reading gets taken though)

This isn't an issue in itself, where it can be bad is when the battery gets low.
If the software is 0.1-0.15V off, that could mean that the battery could get down too 2.85V before the software turns off, possibly severely damaging the battery and shortening its life by quite a bit.

I couldn't find anywhere to calibrate the voltage, so for now, I have just raised the CRIT_LOW_BATTERY_VOLTAGE to 3.2V (and all the voltages above it).

My suggestion is to put a setting in the config.ccp where you calibrate the voltage?

lmarzen, 2024-08-05:
Great observation. The voltage is taken as soon as the esp32 wakes up, before the epaper refreshes.
Also, note that the display will stop refreshing once LOW_BATTERY_VOLTAGE (3.2v) is reached. Once CRIT is reached the esp32 will no longer wake up to check the battery voltage to see if it is okay to refresh again and a manual press of the reset button will be required once the battery is charged again.

I will consider adjusting these to be slightly more conservative.

Asdf1qaz, 2024-08-06:
So the hardware is still charging, even though the voltage is displaying 4.28V, which to me, means that the hardware is reading correctly.

So either:
1. The firebeetle firmware is passing the wrong voltage to the software (I will do some testing tomorrow when I get back home with a small script and hopefully rule this out)

2. Your software is not using the passed voltage correctly and adding an offset

3. Your software is working as intended (which seems most probable as there is no high voltage warning popping up) and an offset that is added after all the calcs and checks are done. Messing with the a calibration or adding an offset would not help in this case as it means it would be changing the correct voltage. Tomorrow, I will confirm this is the issue by connecting the battery input pins to a lab power supply to see if a high voltage warning is displayed when the battery input voltage goes over 4.2V

lmarzen, 2024-08-06:
I think the voltage is read correctly as is. The esp32 reads it from an ADC pin and uses the factory voltage calibration fuses to correctly calculate the corresponding voltage. I believe you are observing the voltage drop that occurs when the epaper screen is refreshing (which occurs after the voltage is measured).

closed 2024-08-25 by lmarzen