lmarzen/esp32-weather-epd #122: When unable to start due to various errors

PaekHyun, 2024-08-19:
Current your code wakes up from sleep mode every hour.
Even if information is not displayed due to various errors, it wakes up from sleep mode every hour.
So what about code that goes into sleep mode for only 2-3 minutes when an error occurs?
I think it's better to wait 2-3 minutes and try to reboot than to see a Wi-Fi connection error message for an hour

 // START WIFI
  int wifiRSSI = 0; 
  wl_status_t wifiStatus = startWiFi(wifiRSSI);
  if (wifiStatus != WL_CONNECTED)
  {
    killWiFi();
    initDisplay();
    if (wifiStatus == WL_NO_SSID_AVAIL)
    {
      Serial.println("SSID Not Available");
      do
      {
        drawError(wifi_x_196x196, "SSID Not Available", "");
      } while (display.nextPage());
    }
    else
    {
      Serial.println("WiFi Connection Failed");
      do
      {
        drawError(wifi_x_196x196, "WiFi Connection", "Failed");
      } while (display.nextPage());
    }
    display.powerOff();
    **esp_sleep_enable_timer_wakeup(180 * 1000000ULL);
    esp_deep_sleep_start();**    
  }
  
  // FETCH TIME
  bool timeConfigured = false;
  timeConfigured = setupTime(&timeInfo);
  if (!timeConfigured)
  { // Failed To Fetch The Time
    Serial.println("Failed To Fetch The Time");
    killWiFi();
    initDisplay();
    do
    {
      drawError(wi_time_4_196x196, "Failed To Fetch", "The Time");
    } while (display.nextPage());
    display.powerOff();

    **esp_sleep_enable_timer_wakeup(180 * 1000000ULL);
    esp_deep_sleep_start();**
  }
  String refreshTimeStr;
  getRefreshTimeStr(refreshTimeStr, timeConfigured, &timeInfo);

  // MAKE API REQUESTS
  int rxOWM[2] = {};
  WiFiClient client;
  rxOWM[0] = getOWMonecall(client, owm_onecall);
  if (rxOWM[0] != HTTP_CODE_OK)
  {
    statusStr = "One Call " + OWM_ONECALL_VERSION + " API";
    tmpStr = String(rxOWM[0], DEC) + ": " + getHttpResponsePhrase(rxOWM[0]);
    killWiFi();
    initDisplay();
    do
    {
      drawError(wi_cloud_down_196x196, statusStr, tmpStr);
    } while (display.nextPage());
    display.powerOff();

    **esp_sleep_enable_timer_wakeup(180 * 1000000ULL);
    esp_deep_sleep_start();**
  }
  rxOWM[1] = getOWMairpollution(client, owm_air_pollution);
  killWiFi(); // wifi no longer needed
  if (rxOWM[1] != HTTP_CODE_OK)
  {
    statusStr = "Air Pollution API";
    tmpStr = String(rxOWM[1], DEC) + ": " + getHttpResponsePhrase(rxOWM[1]);
    initDisplay();
    do
    {
      drawError(wi_cloud_down_196x196, statusStr, tmpStr);
    } while (display.nextPage());
    display.powerOff();

    **esp_sleep_enable_timer_wakeup(180 * 1000000ULL);
    esp_deep_sleep_start();**
  }

Asdf1qaz, 2024-08-30:
This could kill the battery pretty fast though if your not with it

PaekHyun, 2025-05-27:
> It did kill the battery. Maybe you can press reset if you can't wait. I extend the reset button out of the frame.

when you see a message that wifi is not connected or an error message that you can not receive the weather, you can wait an hour or press the reset button.
I'm uncomfortable with those two actions, and the error message itself doesn't come out often, so the battery doesn't disappear particularly quickly.

Asdf1qaz, 2025-05-28:
First off, your title is quite puzzling. In future, please consider a better name like "Decrease refresh time on Wi-Fi connection errors"

Secondly, if your not able to press the reset button when the internet/Wifi has been fixed after a dropout, you should create your own fork and make the changes you require there.

The way @lmarzen has set it up this marvelous project, works for the majority of people.

Say if your Wifi or internet drops at 09:00 and your at work until 17:00. With 2 minute re-polls, it will be making 240 re-try's before you catch it, instead of the regular 16 polls (if kept at the default 30 min refresh interval).
If you have a smaller battery, it will fully drain it.

If your issues are with older network hardware creating problems however, consider increasing the WIFI_TIMEOUT, HTTP_CLIENT_TCP_TIMEOUT and potentially the NTP_TIMEOUT.

Recommending this issue to be closed.

closed 2025-05-28 by lmarzen