lmarzen/esp32-weather-epd #123: Display Not Working - Compatibility with Good Display GDEM075T42?

Leothepig, 2024-09-01:
First, thanks for this great project and your detailed instructions! 

I'm struggling to get my display to work. Serial monitor indicates that it's connecting to WiFi, updating the weather information, then successfully reading from the BME280. Right after that, I get the following:

Busy Timeout!
_PowerOn : 10001247
Busy Timeout!
_Update_Full : 10001135
Busy Timeout!
_PowerOff : 10001107
Awake for 39.726s
Entering deep sleep for 393s
ets Jul 29 2019 12:21:46

I've triple checked my wiring (even resoldering them all once) and the code, and can't find any issue. I'm using all the specific pieces recommended in the guide, except for the screen: I have a black and white 7.5" Good Display [GDEM075T42](https://www.good-display.com/product/581.html) instead of the recommended [GDEY075T7](https://www.good-display.com/product/396.html). Is there any reason this model wouldn't be compatible? The differences seem to be minimal.

The only partial success I've found is when I switch the RESE on the DESPI-C02 from .47 to 3 (on a whim after reading through closed issues)-- I'll get a couple of brief flashes of the correct information displayed, but then it turns to display all black after. I connected the screen to a Raspberry Pi and ran Good Display's test, and it seemed to be working fine.

Any help or advice is much appreciated! I'm happy to add any additional information or pictures if that's helpful.

Leothepig, 2024-09-02:
I changed the 3.3v connection on the DESPI-C02 from D3/26 on the ESP-32-E to 3.3v. The screen comes on now, but flickers a lot and eventually fades to black. My serial monitor data is similar to before, with the flickering and fade to black continuing after it displays "Entering deep sleep." I'm assuming at this point that it's probably most likely a connection/solder quality issue?

Leothepig, 2024-09-02:
Wow, thank you for your help! I don't know how long it would have taken me to figure that out. Nearly every aspect of this project is 90% new for me. 

The link in your guide for the GDEY075T7 actually directs to the GDEM075T42 on AliExpress, if you'd want to update it! I don't think I was able to find the supported panel when I ordered, so maybe Good Display is replacing it? 

Thanks again!

closed 2024-09-02 by Leothepig

lmarzen, 2024-09-02:
Oh, hmm. Maybe this is a new version of the display? I'll investigate this.

Leothepig, 2024-09-02:
Thanks, I appreciate it! In the meantime, I've ordered the Waveshare screen to be safe.

lmarzen, 2024-09-02:
It's frustrating that they retroactively redirected the page for the original product to an incompatible product. I have updated the readme to remove the link to purchase this panel.

There is a review on the [aliexpress page of the new display](https://www.aliexpress.us/item/3256802771862219.html) from July 18th 2024 that comments on this incompatibility with GxEPD2.

> This display is not compatible with GxEPD2 or GxEPD libraries, so you cant use partial refresh. The example provided by Good Display is very low level and you cant use simple functions such as printing text. The customer service didnt help me. Please dont buy this display.

Leothepig, 2024-09-03:
Aha, thanks for helping put the pieces together! I appreciate your help!

lmarzen, 2024-10-27:
The correct product is back up on the Good Display aliexpress store page. Not sure why they did this. smh