lmarzen/esp32-weather-epd #125: Automatically switch between known wifi networks

Beardslow55, 2024-09-17:
I found and built this project for my wife and she loves it.  I'm still fairly new to programing, ardunio, github and the like but had a question.  On occasion she brings her weather station home for a few days but I have to reflash it to our home network wifi in order for it to work.  Is it possible to set up 2 or 3 "known" wifi SSIDs and passwords and the esp32 automatically try each one in a specified order?  I'd be ok if I had to press the reset button in order for it to "scan" for the known 2 or 3 networks.  Thanks in advance.

lmarzen, 2024-09-17:
This is certainly possible and is a planned feature. No promise on when exactly it will come, I am a busy student, however, I hope to spend some time adding that feature and a bunch more once I graduate in December.

blind-oracle, 2025-02-02:
@lmarzen I also need this feature and probably will implement it anyway, are PRs for that accepted?

lmarzen, 2025-02-02:
Glad to accept PRs :)

blind-oracle, 2025-02-02:
Okey dockey here it goes https://github.com/lmarzen/esp32-weather-epd/pull/155