lmarzen/esp32-weather-epd #127: clarifications around e-paper display-HAT compatibility + refresh rate

mmacchia, 2024-09-28:
Hello!

First of all thank you for bringing this exciting project to life.

I would like to ask for some clarifications:
* Are the waveshare and goodisplay screens interchangeable? In other words, is Waveshare 13187 compatible with DESPI-C02 Adapter Board?
* Regarding the sleep duration, you mention in the code comment
```
// Sleep duration in minutes. (aka how often esp32 will wake for an update)
// Aligned to the nearest minute boundary and must evenly divide 60.
// For example, if set to 30 (minutes) the display will update at 00 or 30
// minutes past the hour. (range: [2-60])
```
what prevents to set it to >60, e.g 120, 180?

Thank you for your support :)

lmarzen, 2024-09-28:
Yes, the goodisplay and waveshare panels are interchangeable. Infact, the configuration I am currently using is the recommended Waveshare panel with the DESPI.

I'll double check on the sleep parameter and make sure there aren't any bugs for times greater than 60 minutes. I think it should work fine, but I probably put that range because I never probably tested it or there may be a small bug that I was aware of at the time but have since forgotten 😅.

lmarzen, 2024-09-28:
I fixed the bug that would occur for values not do not evenly divide 60. The SLEEP_DURATION option is now much more robust. 
I tested it for a while so hopefully, there are now glaring issues with it.

closed 2024-09-28 by lmarzen

mmacchia, 2024-09-29:
Thank you for looking into this and implementing the fix so quickly!
I will test the code as soon as I receive the e-paper screen :)