lmarzen/esp32-weather-epd #128: E-paper display power consumption

androot2, 2024-10-24:
Are you disconnecting EPD power supply by GPIO pin? Because after that, display need to be full refreshed right?
I have similiar project, after ESP32 deep sleep I refreshing only minutes part on display. But I can't disconnect power supply, because partial refresh won't work. EPD power consumption, even after display.hibernate(); (GxEPD2_BW library) is too big for battery operation ~0,5mA measured in VCC line to EPD. My display is smaller, WeAct Studio, 2.13. It could have lower power consumption according to data sheet. Do you have any idea how to solve that?

mgmomo, 2024-10-24:
Hi, it can be handled by an "ugly" trick.  I created a similar project and I am disconnection the power suply with a mos fet. But I connect drain and source with a 22k ohm resistor. In my project the current is reduced. I dont remember the total numbers but I think it are about 50 to 70 uA. With an 2800 mAh Accu, the system is available for about 3 to 4 weeks. The system is in deep sleep for 1 minute. Every 10 minutes a full refresh is required. 

[image]

androot2, 2024-10-25:
Hello, thank you for answer. I found a solution on Internet. Weak pull-up on E-paper RESET pin. I added 220 k resistor to 3.3 V. Current consumption drops to data sheet value. I think it could help for others. now E-paper Vcc pin may be connected directly to 3.3 V.

closed 2024-10-25 by androot2

patcher-ms, 2025-04-02:
@androot2 I also have a WeAct Studio display (4.2" B/W) and I'm seeing a >500uA current when the display is in deep sleep.

From your comment I understand that you are connecting all the display pins directly to the ESP32 pins except for `reset`, where you are using a 220k pull-up resistor to 3.3V. Is this correct? I just tried the same and for me it didn't reduce the power consumption.

androot2, 2025-04-03:
EPD RST is connected to ESP32 pin and added external pullup 100-220k (doesn't matter).
EPD Vcc directly to power supply.

patcher-ms, 2025-04-03:
Thank you for your reply! That sounds pretty much what I tried but I can give it another try later.

A difference is that I was connecting EPD VCC to the 3.3V pin of the ESP32. I can try to connect it straight to the power supply. https://github.com/WeActStudio/WeActStudio.EpaperModule/issues/12#issuecomment-1851962229 shows that using 5V instead of 3.3V makes a big difference. My 3.3V current is 10x what they’re reporting but it’s definitely worth a try

androot2, 2025-04-03:
I don't use any of WeAct Studio boards, so I don't know. I have bare ESP32 S3 chip and some other components to drive EPD properly. I prepared PCBs for ESP32S3 and ESP32C3 (single core, less power supply). EPD RST pull up is very important to proper go to deep sleep mode. If you don't have it, EPD is woke up immediatelly and current is >500 uA.

[weather2.pdf](https://github.com/user-attachments/files/19584014/weather2.pdf)

Schematic is prepared for WROOM modules, and bare chips C3 and S3 so don't be confused with so many ICs :)

patcher-ms, 2025-04-03:
Earlier I chatted with WeAct Studio and they told me to connect VCC of the EPD to a source > 3.3V. I tried it and the difference is measurable. I haven't had a chance to look into why.

The current consumption of both Xiao ESP32C3 and the 4.2" display with both on deep sleep changed from ~650uA when I connected VCC of the EPD to the 3V3 pin on the board to ~105uA when I connected the EPD VCC to the power supply.