lmarzen/esp32-weather-epd #129: One call API Upgrade 2.5 to 3.0

Ianszh, 2024-10-29:
Hello Luke,

Thanks again for the great project, it's been running fine for ages but now though I see that the 2.5 APi has been deprecated.

I've created a new 3.0 subscription and limited it to 1000 call per day, I take it I'll have to reinstall the whole project to change the API call?

Thanks Ian

lmarzen, 2024-10-29:
Yes, you are correct.

lmarzen, 2024-10-29:
I think in theory it may be possible, though it would be non-trivial and highly technical. Unfortunately, you'll have to start clean, unless you can find your old files.

Side note, I am slowly working on a web interface for this project in the background, which would not require a reflash to update settings. Hopefully, one day, eliminating some of the user-friendliness issues.

Ianszh, 2024-10-29:
I thought so, I'll try and find the old file or start again.

That would be cool, I see a few other projects going / gone that way.

lmarzen, 2024-10-29:
godspeed

closed 2024-10-29 by lmarzen