lmarzen/esp32-weather-epd #130: Too many request on second station

miszowaty77, 2024-12-06:
I really liked the project, so I decided to make a second one as a gift. I uploaded the same firmware to both esp and to my surprise, both now shows the message "too many requests". I set the limit to 999. Could this be a problem with synchronizing both units at the same time?

lmarzen, 2024-12-07:
Sounds like you are hitting the daily rate limit.

What is your SLEEP_DURATION set to?

You would expect to exhaust 1000 calls a day if you updated every 1.44 minutes with no failed request. for 2 displays with no api call retries each display at 2.88 minutes per call would exhaust.

The display will retry API calls 3 times before giving up and displaying an error. In the worst case, assuming 3 retries per refresh per display, each display would exhaust the daily limit if set to a SLEEP_DURATION of < 9.

Go to this link to see your api call usage. https://home.openweathermap.org/statistics/onecall_30

miszowaty77, 2024-12-07:
[image]

I uploaded the exact same firmware to both units yesterday around 7pm. For some reason all queries ran out by 9pm.

Refresh time is default = 30min.

lmarzen, 2024-12-07:
Hmmm. That's interesting.

Is the issue still persisting?

If it is, it would be helpful if you could connect a usb to one of them and check what is being printed to the terminal output. Copy and paste that here.

miszowaty77, 2024-12-17:
This bug never were never repeated.

closed 2024-12-17 by miszowaty77