lmarzen/esp32-weather-epd #135: UV index not showing

zijing99, 2024-12-20:
It seems like there might be a bug or a misconfiguration. After uploading the code to the ESP32 and sending a GET request to the API, the UVI value is not returned, and it shows 0 for the current UV index. I set the log level to 2 and monitored the serial output, where the current UVI is also reported as 0. However, when I send the same GET request via Postman, the API successfully returns the correct UVI value. 

Double check the timestamp is matching. 

Do anyone notice UV index not showing?

lmarzen, 2024-12-20:
I can't say I have noticed this before. Maybe the location you're giving is different?
I double-checked that they didn't change the response format. I thought that could have been a cause for this issue, but it is unchanged.

zijing99, 2024-12-20:
I am sending the exact query from the serial monitor using Google Chrome

https://api.openweathermap.org/data/3.0/onecall?lat=5.389175&lon=100.416464&lang=en&units=standard&exclude=minutely&appid=<apikey>

Google Chrome Response 
{current":"dt":1734688185,"sunrise":1734650500,"sunset":1734693012,"temp":304.13,"feels_like":309.54,"pressure":1007,"humidity":66,"dew_point":297.03,"uvi":0.69

Serial Monitor Showing
{"current":"dt":1734688189,"sunrise":1734650500,"sunset":1734693012,"temp":304.13,"feels_like":309.54,"pressure":1007,"humidity":66,"dew_point":297.03,"uvi":0

Is there any data preprocessing or cleaning after retrieving the raw JSON from the API?"

lmarzen, 2024-12-20:
Ahhh, maybe it's a rounding issue.
let me check

lmarzen, 2024-12-20:
Hmmm, when I use the same coordinates on my unit, I get the expected uvi of 1 to show up.

lmarzen, 2024-12-20:
Try adding some print statements in renderer.cpp at line 455. This should help us know exactly where the issue is happening for you.

```C++
  ...
  display.setFont(&FONT_12pt8b);
  uint uvi = static_cast<uint>(std::max(std::round(current.uvi), 0.0f));
  // vvv
  Serial.println("float UVI: " + String(current.uvi));
  Serial.println("round UVI: " + String(std::round(current.uvi)));
  Serial.println("max UVI: " + String(std::max(std::round(current.uvi), 0.0f)));
  Serial.println("uint UVI: " + String(uvi));
  // ^^^
  dataStr = String(uvi);
  drawString(48, 204 + 17 / 2 + (48 + 8) * 2 + 48 / 2, dataStr, LEFT);
  ...
  ```

zijing99, 2024-12-20:
I have pull down a fresh copy from the repo, however the UV Index is still not showing, here is the Serial output

float UVI: 0.00
round UVI: 0.00
max UVI: 0.00
uint UVI: 0

lmarzen, 2024-12-20:
Wow, it really is just 0.
Maybe it's an issue with the json parsing or maybe OWM doesnt like your esp32 or maybe it's something about the api call?
Im at a loss, this is unexpected.

lmarzen, 2024-12-20:
I just tried it on my phone in firefox and i got 0, but from my laptop on firefox I got 0.6

zijing99, 2024-12-20:
Seems like something is weird with the api call

zijing99, 2024-12-22:
After some investigation, I found that there’s a problem with how the OpenWeatherMap API is providing UVI data. It always show 0 in my current location although other weather provider is reporting 10. I’ve already reached out to OpenWeatherMap support to report this issue and am waiting for their response.

closed 2024-12-22 by zijing99