lmarzen/esp32-weather-epd #138: No output.

guus6457, 2024-12-28:
Hello I did every single step and there is no display [image]

lmarzen, 2024-12-28:
Hello,

Sorry to hear it's not working. Please share the terminal output so I can better help you. It's hard to help without additional information.

In the picture it looks like your display may not be plugged all the way into your driver board.

guus6457, 2024-12-29:
Hi,

There is no output that's the problem I want to do a hard reset but I can't

guus6457, 2024-12-29:
and it saying Leaving...
Hard resetting via RTS pin...  and then notting happens

lmarzen, 2024-12-29:
Is the code successfully uploading to the microcontroller?
There should be a serial terminal output over usb. you can view it by going to the serial monitor tab in vscode/platformio (along the bottom near the upload button).

lmarzen, 2024-12-29:
Did you try reseating the ribbon cable from the display into the driver board? You need to lift the black lever before you can insert the cable all the way, then push the black level back down.

guus6457, 2025-01-01:
hi, okay let me try it

guus6457, 2025-01-01:
There is no output

lmarzen, 2025-01-01:
What panel do you have? Did you select the correct option in config.h corresponding to thr panel you are using?

guus6457, 2025-01-01:
Waveshare 7.5 inch E-Ink E-Paper Display
--

<br class="Apple-interchange-newline">

lmarzen, 2025-01-01:
There are multiple different 7.5in panels, you need to find out the resolution and hoe many colors yours supports so you can select the correct panel in config.h.

guus6457, 2025-01-01:
okay get it where can i change it i have the Waveshare 7.5 inch E-Ink E-Paper Display. black and white

guus6457, 2025-01-01:
okay i upload the code it work but it dont display on the on the display

lmarzen, 2025-01-06:
Please share the terminal output so I can better help you. Click the 'Terminal Monitor' button in platformio/vscode (at the bottom near the Upload button). Then press the reset button on the esp32 and it should print to the terminal. Copy and paste the output here (do not post a screenshot of the output as this make it difficult to search).

closed 2026-01-05 by lmarzen