lmarzen/esp32-weather-epd #139: No display output

NathanKolbas, 2024-12-29:
My apologies if this is not the right place to put this.

I am running into an issue where I am getting no display output. I am using the FireBeetle 2 ESP32-E linked in the README and the Good Display 7.5in also linked in the README. I tried two ESP32s. The first one I soldered the wires directly to the ESP32 and the second one is attached via breadboard. I tested all the connections with a multimeter to make sure the wires were connected. I also tried to use the 3.3v rail. The switch on the DESPI-CO2 is set to 0.47 (tried flipping it a few times). Double checked that the display ribbon cable is connected to the DESPI-CO2 (I can see indents on the ribbon cable's copper).

Here are some photos:
[image]
[image]
[image]
[image]

I am using the latest version of the repo on the main branch ([c7b343d](https://github.com/lmarzen/esp32-weather-epd/commit/c7b343ddbea77a5acb12a5d2197948c79bec0e12)). The only thing changed was `config.cpp`  with the usual (I removed the API key as to not make unneeded calls as it should still update the display).

Here is the output from serial after pressing the reset button a few times (some output removed that is not needed):
```
Reconnecting to COM7     Connected!
............
_PowerOn : 26
Busy Timeout!
_Update_Full : 10001131
Busy Timeout!
_PowerOff : 10001115
Awake for 33.738s
Entering deep sleep for 878s

rst:0x1 (POWERON_RESET),boot:0x13 (SPI_FAST_FLASH_BOOT)
configsip: 0, SPIWP:0xee
clk_drv:0x00,q_drv:0x00,d_drv:0x00,cs0_drv:0x00,hd_drv:0x00,wp_drv:0x00
mode:DIO, clock div:2
load:0x3fff0030,len:1184
load:0x40078000,len:13232
load:0x40080400,len:3028
entry 0x400805e4
Battery voltage: 4258mv
_PowerOn : 25
Busy Timeout!
_Update_Full : 10001131
Busy Timeout!
_PowerOff : 10001115
Awake for 46.522s
Entering deep sleep for 827s

rst:0x1 (POWERON_RESET),boot:0x13 (SPI_FAST_FLASH_BOOT)
configsip: 0, SPIWP:0xee
clk_drv:0x00,q_drv:0x00,d_drv:0x00,cs0_drv:0x00,hd_drv:0x00,wp_drv:0x00
mode:DIO, clock div:2
load:0x3fff0030,len:1184
load:0x40078000,len:13232
load:0x40080400,len:3028
entry 0x400805e4
Battery voltage: 4128mv
_PowerOn : 26
Busy Timeout!
_Update_Full : 10001131

rst:0x1 (POWERON_RESET),boot:0x13 (SPI_FAST_FLASH_BOOT)
configsip: 0, SPIWP:0xee
clk_drv:0x00,q_drv:0x00,d_drv:0x00,cs0_drv:0x00,hd_drv:0x00,wp_drv:0x00
mode:DIO, clock div:2
load:0x3fff0030,len:1184
load:0x40078000,len:13232
load:0x40080400,len:3028
entry 0x400805e4
Battery voltage: 4284mv
_PowerOn : 26
Busy Timeout!
_Update_Full : 10000131
Busy Timeout!
_PowerOff : 10000115
Awake for 31.013s
Entering deep sleep for 724s

rst:0x5 (DEEPSLEEP_RESET),boot:0x13 (SPI_FAST_FLASH_BOOT)
configsip: 0, SPIWP:0xee
clk_drv:0x00,q_drv:0x00,d_drv:0x00,cs0_drv:0x00,hd_drv:0x00,wp_drv:0x00
mode:DIO, clock div:2
load:0x3fff0030,len:1184
load:0x40078000,len:13232
load:0x40080400,len:3028
entry 0x400805e4
Battery voltage: 4210mv
Busy Timeout!
_PowerOn : 10001191
Busy Timeout!
_Update_Full : 10001123
Busy Timeout!
_PowerOff : 10001115
Awake for 40.385s
Entering deep sleep for 1771s
Disconnected
```

Any suggestions on what I could do or anything that looks incorrect? Thank you for your time and help!

samlhouston, 2024-12-29:
Looking at you pictures, it looks like your display ribbon code is upside down. I made the same mistake.
Try flipping it. Best of luck.
[image]

NathanKolbas, 2024-12-29:
@samlhouston That was was my problem. Can't believe I missed that... Thank you so much!

closed 2024-12-29 by NathanKolbas