lmarzen/esp32-weather-epd #140: Display is not turning on, do i need a battery?

dorofino, 2024-12-30:
The display is not turning. Do I need the battery connected, or can I test it with just the USB connected? I believe I connected everything correctly, but the display is still not turning on. I see the battery LED blinking on the board, so perhaps it is due to the absence of the battery. Any advice on this?

[image]

lmarzen, 2024-12-30:
No battery needed.

Please share the serial monitor output. (go to the Serial Monitor tab at the bottom of vscode/platformio near the upload button)

dorofino, 2024-12-30:
Here

```
--- Terminal on COM7 | 115200 8-N-1
--- Available filters and text transformations: colorize, debug, default, direct, esp32_exception_decoder, hexlify, log2file, nocontrol, printable, send_on_enter, time
--- More details at https://bit.ly/pio-monitor-filters
--- Quit: Ctrl+C | Menu: Ctrl+T | Help: Ctrl+T followed by Ctrl+H
Busy Timeout!
_Update_Full : 10000150
Busy Timeout!
_PowerOff : 10001107
Awake for 41.290s
Entering deep sleep for 1737s
ets Jul 29 2019 12:21:46

rst:0x1 (POWERON_RESET),boot:0x13 (SPI_FAST_FLASH_BOOT)
configsip: 0, SPIWP:0xee
clk_drv:0x00,q_drv:0x00,d_drv:0x00,cs0_drv:0x00,hd_drv:0x00,wp_drv:0x00
mode:DIO, clock div:2
load:0x3fff0030,len:1184
load:0x40078000,len:13232
load:0x40080400,len:3028
entry 0x400805e4
Battery voltage: 4170mv
Connecting to 'UNBREAKABLE-NETWORK'...
IP: 192.168.1.220
Waiting for SNTP synchronization........................................
Monday, December 30, 2024 12:39:00
Attempting HTTP request: api.openweathermap.org/data/3.0/onecall?lat=30.0550818&lon=-81.5166671&lang=en&units=standard&exclude=minutely&appid={API key}
  200 OK
Attempting HTTP request: api.openweathermap.org/data/2.5/air_pollution/history?lat=30.0550818&lon=-81.5166671&start=1735493943&end=1735580342&appid={API key}
  200 OK
Reading from BME280... Success
Busy Timeout!
_PowerOn : 10001255
Busy Timeout!
_Update_Full : 10000150
Busy Timeout!
_PowerOff : 10001107
Awake for 40.217s
Entering deep sleep for 1227s
```

I also connected the display directly to the DESPI-C02, and I believe I have the wiring correctly matching your wiring diagram/wiring_di_despi-c02.png.

[image]
[image]

dorofino, 2024-12-30:
Additionally, I observe voltage the DESPI-C02 for a while after bootup and after invoking the api.openweathermap.

[image]

[image]

lmarzen, 2024-12-30:
It looks like the connector on your epaper display is not all the way plugged into the driver board? (pull the black lever up, insert the cable (shouldn't be much resistance), then push the black lever back down.).

dorofino, 2024-12-30:
Yes, that is what I thought, but can see bite marks in the picture.
[image]

[image]

The display is a brand-new display.

lmarzen, 2024-12-30:
I must be mistaken then. I'll take a closer look at your wiring when I get a chance.

dorofino, 2024-12-30:
Thank you. I over the wiring a couple of times and I they are ok.

dorofino, 2024-12-31:
I purchase this E-ink Display, perhaps this is the issue? Do you think is not compatible?

https://www.aliexpress.us/item/3256806387212613.html?spm=a2g0o.order_list.order_list_main.28.21ef18022RnFjm&gatewayAdapt=glo2usa#nav-specification

lmarzen, 2024-12-31:
Did you select the corresponding panel in config.h?

dorofino, 2024-12-31:
I believed i did, I follow the instructions. 

here is the config.h

```
/* Configuration option declarations for esp32-weather-epd.
 * Copyright (C) 2022-2024  Luke Marzen
 *
 * This program is free software: you can redistribute it and/or modify
 * it under the terms of the GNU General Public License as published by
 * the Free Software Foundation, either version 3 of the License, or
 * (at your option) any later version.
 *
 * This program is distributed in the hope that it will be useful,
 * but WITHOUT ANY WARRANTY; without even the implied warranty of
 * MERCHANTABILITY or FITNESS FOR A PARTICULAR PURPOSE.  See the
 * GNU General Public License for more details.
 *
 * You should have received a copy of the GNU General Public License
 * along with this program.  If not, see <https://www.gnu.org/licenses/>.
 */

#ifndef __CONFIG_H__
#define __CONFIG_H__

#include <cstdint>
#include <Arduino.h>

// E-PAPER PANEL
// This project supports the following E-Paper panels:
//   DISP_BW_V2 - 7.5in e-Paper (v2)      800x480px  Black/White
//   DISP_3C_B  - 7.5in e-Paper (B)       800x480px  Red/Black/White
//   DISP_7C_F  - 7.3in ACeP e-Paper (F)  800x480px  7-Color
//   DISP_BW_V1 - 7.5in e-Paper (v1)      640x384px  Black/White
// Uncomment the macro that identifies your physical panel.
#define DISP_BW_V2
// #define DISP_3C_B
// #define DISP_7C_F
// #define DISP_BW_V1

// E-PAPER DRIVER BOARD
// The DESPI-C02 is the only officially supported driver board.
// Support for the Waveshare rev2.2 and rev2.3 is deprecated.
// The Waveshare rev2.2 is no longer in production.
// Users of the Waveshare rev2.3 have reported experiencing low contrast issues.
// Uncomment the macro that identifies your driver board hardware.
#define DRIVER_DESPI_C02
// #define DRIVER_WAVESHARE

// 3 COLOR E-INK ACCENT COLOR
// Defines the 3rd color to be used when a 3+ color display is selected.
#if defined(DISP_3C_B) || defined(DISP_7C_F)
  // #define ACCENT_COLOR GxEPD_BLACK
  #define ACCENT_COLOR GxEPD_RED
  // #define ACCENT_COLOR GxEPD_GREEN
  // #define ACCENT_COLOR GxEPD_BLUE
  // #define ACCENT_COLOR GxEPD_YELLOW
  // #define ACCENT_COLOR GxEPD_ORANGE
#endif

// LOCALE
// If your locale is not here, you can add it by copying and modifying one of
// the files in src/locales. Please feel free to create a pull request to add
// official support for your locale.
//   Language (Territory)            code
//   German (Germany)                de_DE
//   English (United Kingdom)        en_GB
//   English (United States)         en_US
//   Estonian (Estonia)              et_EE
//   Finnish (Finland)               fi_FI
[... 310 lines cut ...]
 && !(  defined(WIND_ICONS_CARDINAL)                \
      ^ defined(WIND_ICONS_INTERCARDINAL)           \
      ^ defined(WIND_ICONS_SECONDARY_INTERCARDINAL) \
      ^ defined(WIND_ICONS_TERTIARY_INTERCARDINAL)  \
      ^ defined(WIND_ICONS_360))
  #error Invalid configuration. Exactly one wind direction icon precision level must be selected.
#endif
#if !(defined(FONT_HEADER))
  #error Invalid configuration. Font not selected.
#endif
#if !(defined(DISPLAY_DAILY_PRECIP))
  #error Invalid configuration. DISPLAY_DAILY_PRECIP not defined.
#endif
#if !(defined(DISPLAY_ALERTS))
  #error Invalid configuration. DISPLAY_ALERTS not defined.
#endif
#if !(defined(BATTERY_MONITORING))
  #error Invalid configuration. BATTERY_MONITORING not defined.
#endif
#if !(defined(DEBUG_LEVEL))
  #error Invalid configuration. DEBUG_LEVEL not defined.
#endif

#endif
```

lmarzen, 2024-12-31:
The link you sent is actually to a 3 color panel.
Try using
```C
#define DISP_3C_B
```

dorofino, 2025-01-01:
I tried that, and still the same issue persists.
At one point, and I don't know how/why, the screen turned like this.

[image]

and I finally installed the battery, and the same issue remains.

[image]

I am wondering is a problem with the display.

lmarzen, 2025-01-01:
Probably not an issue with the panel itself. Most likely a wiring or software problem.

Try flipping the switch on the despi back and forth once. In the past some people have had similar issues when the switch appeared to be set all the way, and resetting it fixed it.

Try search some of the other issues if you haven't already. You will find some other similar problems with troubleshooting ideas and solutions.

dorofino, 2025-01-02:
Could it be any of these settings? what do you mean 'micro-usb firebeetle'

```
// ADC pin used to measure battery voltage
const uint8_t PIN_BAT_ADC  = A2; // A0 for micro-usb firebeetle
// Pins for E-Paper Driver Board
const uint8_t PIN_EPD_BUSY = 14; // 5 for micro-usb firebeetle
const uint8_t PIN_EPD_CS   = 13;
const uint8_t PIN_EPD_RST  = 21;
const uint8_t PIN_EPD_DC   = 22;
const uint8_t PIN_EPD_SCK  = 18;
const uint8_t PIN_EPD_MISO = 19; // 19 Master-In Slave-Out not used, as no data from display
const uint8_t PIN_EPD_MOSI = 23;
const uint8_t PIN_EPD_PWR  = 26; // Irrelevant if directly connected to 3.3V
// I2C Pins used for BME280
const uint8_t PIN_BME_SDA = 17;
const uint8_t PIN_BME_SCL = 16;
const uint8_t PIN_BME_PWR =  4;   // Irrelevant if directly connected to 3.3V
const uint8_t BME_ADDRESS = 0x76; // If sensor does not work, try 0x77
```

lmarzen, 2025-01-02:
If you are using an older version of the dfrobot firebeetle with a micro usb connector instead of the newer usb c version.

dorofino, 2025-01-03:
no, i have the new firebeetle.

Quick question: How I go about testing other option/drivers and debugging see how can make it work?

lmarzen, 2025-01-03:
This project uses the [GxEPD2](https://github.com/ZinggJM/GxEPD2) driver library. There are some minimal hello world examples there if you follow the link. Else there are other driver libraries to, but I have personally not used them much.

lmarzen, 2025-01-03:
The thick boarder around the edge of the screen is sometimes a symptom of low voltage in my experience. Maybe try connecting the wire that goes to the driver board's 3.3v pin directly to a 3.3v pin on the esp32.

lmarzen, 2025-01-03:
If you want to mess around with the display initialization code it is in renderer.cpp:
https://github.com/lmarzen/esp32-weather-epd/blob/c7b343ddbea77a5acb12a5d2197948c79bec0e12/platformio/src/renderer.cpp#L223C1-L250C21

Maybe try moving the spi remap before the display init?

Like this...
```C++
  // remap spi
  SPI.end();
  SPI.begin(PIN_EPD_SCK,
            PIN_EPD_MISO,
            PIN_EPD_MOSI,
            PIN_EPD_CS);
display.init(115200, true, 10, false);
```

Otherwise maybe you can try different init parameters from the examples in GxEPD2.

dorofino, 2025-01-03:
> The thick boarder around the edge of the screen is sometimes a symptom of low voltage in my experience. Maybe try connecting the wire that goes to the driver board's 3.3v pin directly to a 3.3v pin on the esp32. 

So I believe I did it right, and I connected the DESPI-C02 pin 3.3v directly to that ESP32-E V1.0 pin 3V3

[image]

I also moved the SPI

```
  // remap spi
  SPI.end();
  SPI.begin(PIN_EPD_SCK,
            PIN_EPD_MISO,
            PIN_EPD_MOSI,
            PIN_EPD_CS);
  display.init(115200, true, 10, false);
```

And now after resetting I see this

https://github.com/user-attachments/assets/559b0d17-5c21-4247-805e-b9b6e0ec3d55

lmarzen, 2025-01-05:
Try changing the reset duration from 10 to 20 in display.init. I think that might fix your issue.

```C++
display.init(115200, true, 20, false);
```

veganhamster, 2025-01-20:
I had a similiar problem - maybe my solution is some idea for you?

I also started the project and had (with a b/w screen) only no-screen and in the end on disturbed line at the bottom. The code was correct installed on the esp32 and i can see, it loggged in correctly over the internet at openwather. Hmm.

So i remembered, at the beginning, I connected the battery and AFTER that, i recognize the polarity was wrong. Oops.

So, I changed the FireBeetle ESP with a new one, this time it worked like a charme... So maybe you can try this solution?

blind-oracle, 2025-02-02:
In my case the display doesn't work either if I connect its power to some random pin (26 by default). If I switch it to 3.3v then all works fine.

I've tried setting the pin current capability higher using https://docs.espressif.com/projects/esp-idf/en/stable/esp32/api-reference/peripherals/gpio.html#_CPPv429rtc_gpio_set_drive_capability10gpio_num_t16gpio_drive_cap_t but that didn't help either.

lmarzen, 2025-03-24:
Hi @freakydude 

I updated the readme and wiring diagram to show that the display should be connect to 3.3v instead of the gpio pin. (e9193b3008fa0738c019364b1b3bcfda0b55726d).

It should be possible to do partial refreshes so long as power to the display is maintained. I've thought about implementing. I just haven't gotten around to it as the full refresh doesn't bother me. Patches are welcome.

freakydude, 2025-03-24:
@lmarzen Sorry, I didn't see you updated this 2 days ago.

Thanks for the clarification. Yes, I'm not sure yet if I'll take the time to improve this, but I'll send you a pull request if I do ;-)
Great work.