lmarzen/esp32-weather-epd #141: Waveshare 7.3inch 6-Color won't display

zombor, 2024-12-30:
I'm trying to get this project working with `Waveshare 7.3inch 6-Color E-Paper Display Module` but the display doesn't seem to come on. It seems to have the same HAT as `Waveshare 7.3in ACeP e-Paper (F)` so I figured it might work.

I changed `include/config.h`:

```
$ git diff include/config.h
diff --git a/platformio/include/config.h b/platformio/include/config.h
index 67eaa6c4..6ca85f1b 100644
--- a/platformio/include/config.h
+++ b/platformio/include/config.h
@@ -28,9 +28,9 @@
 //   DISP_7C_F  - 7.3in ACeP e-Paper (F)  800x480px  7-Color
 //   DISP_BW_V1 - 7.5in e-Paper (v1)      640x384px  Black/White
 // Uncomment the macro that identifies your physical panel.
-#define DISP_BW_V2
+// #define DISP_BW_V2
 // #define DISP_3C_B
-// #define DISP_7C_F
+#define DISP_7C_F
 // #define DISP_BW_V1

 // E-PAPER DRIVER BOARD
@@ -39,8 +39,8 @@
 // The Waveshare rev2.2 is no longer in production.
 // Users of the Waveshare rev2.3 have reported experiencing low contrast issues.
 // Uncomment the macro that identifies your driver board hardware.
-#define DRIVER_DESPI_C02
-// #define DRIVER_WAVESHARE
+// #define DRIVER_DESPI_C02
+#define DRIVER_WAVESHARE

 // 3 COLOR E-INK ACCENT COLOR
 // Defines the 3rd color to be used when a 3+ color display is selected.
```

and hooked up the wires like I've seen in the examples (but this could be wrong and my whole problem):

[image]
[image]

Console debug output:

```
rst:0x1 (POWERON_RESET),boot:0x13 (SPI_FAST_FLASH_BOOT)
configsip: 0, SPIWP:0xee
clk_drv:0x00,q_drv:0x00,d_drv:0x00,cs0_drv:0x00,hd_drv:0x00,wp_drv:0x00
mode:DIO, clock div:2
load:0x3fff0030,len:1184
load:0x40078000,len:13232
load:0x40080400,len:3028
entry 0x400805e4
Battery voltage: 4174mv
...
Reading from BME280... BME not found
_InitDisplay reset : 17
_PowerOn : 1
_refresh : 2938000
_PowerOff : 1
Awake for 11.303s
Entering deep sleep for 1844s
```

I'm trying to get this working without soldering first, is that possible? I'm just using the header pins that came with the board right now. This is my first project like this so apologies if I've done anything dumb!

lmarzen, 2024-12-30:
You need to solder the header pins to the esp32 board in order to make a good electrical connection.

zombor, 2024-12-31:
Thanks, after soldering the pins and moving the grey cable to the 3.3V pin, it works :)

Of course doesn't actually display 6 colors, but I might try and modify the code to do that later. My battery doesn't seem to charge from usb (charge light stays off even when usb plugged in, but I'll try and figure that out on my own.

closed 2025-01-01 by zombor