lmarzen/esp32-weather-epd #142: invalid header: 0xffffffff

zombor, 2025-01-01:
I'm having this problem after deep sleep wakeup, not sure what is causing it. I lowered the sleep time to more reliably reproduce. Sometimes it comes out of deep sleep fine, but it eventually gets stuck in this error loop. Some googling about it suggests that the error means there's no bootloader or partition table. But why would it work okay for a few times and then get stuck in the error loop?

```
...
Awake for 22.942s
Entering deep sleep for 74s
ets Jul 29 2019 12:21:46

rst:0x5 (DEEPSLEEP_RESET),boot:0x13 (SPI_FAST_FLASH_BOOT)
configsip: 0, SPIWP:0xee
clk_drv:0x00,q_drv:0x00,d_drv:0x00,cs0_drv:0x00,hd_drv:0x00,wp_drv:0x00
mode:DIO, clock div:2
load:0x3fff0030,len:1184
load:0x40078000,len:13232
load:0x40080400,len:3028
entry 0x400805e4
[debug] Heap Size       : 259016 B
[debug] Available Heap  : 232092 B
[debug] Min Free Heap   : 226708 B
[debug] Max Allocatable : 110580 B
[debug] ADC Cal eFuse Vref
Battery voltage: 4016mv
...
Waiting for SNTP synchronization.........
Wednesday, January 01, 2025 09:51:02
...
Reading from BME280... BME not found
_InitDisplay reset : 3996
_PowerOn : 173000
[debug] alerts.size()    : 0
[debug] alerts.size()    : 0
[debug] alerts.size()    : 0
[debug] alerts.size()    : 0
_refresh : 14499007
_PowerOff : 146000
[debug] Heap Size       : 256832 B
[debug] Available Heap  : 210492 B
[debug] Min Free Heap   : 105796 B
[debug] Max Allocatable : 110580 B
Awake for 22.625s
Entering deep sleep for 98s
ets Jul 29 2019 12:21:46

rst:0x5 (DEEPSLEEP_RESET),boot:0x13 (SPI_FAST_FLASH_BOOT)
invalid header: 0xffffffff
invalid header: 0xffffffff
invalid header: 0xffffffff
invalid header: 0xffffffff
invalid header: 0xffffffff
invalid header: 0xffffffff
invalid header: 0xffffffff
invalid header: 0xffffffff
invalid header: 0xffffffff
invalid header: 0xffffffff
invalid header: 0xffffffff
invalid header: 0xffffffff
invalid header: 0xffffffff
invalid header: 0xffffffff
ets Jul 29 2019 12:21:46

rst:0x7 (TG0WDT_SYS_RESET),boot:0x13 (SPI_FAST_FLASH_BOOT)
invalid header: 0xffffffff
invalid header: 0xffffffff
invalid header: 0xffffffff
invalid header: 0xffffffff
invalid header: 0xffffffff
invalid header: 0xffffffff
invalid header: 0xffffffff
invalid header: 0xffffffff
invalid header: 0xffffffff
invalid header: 0xffffffff
invalid header: 0xffffffff
invalid header: 0xffffffff
invalid header: 0xffffffff
invalid header: 0xffffffff
ets Jul 29 2019 12:21:46
```

Then it constantly prints the `rst:0x7 (TG0WDT_SYS_RESET),boot:0x13 (SPI_FAST_FLASH_BOOT)` stuff over and over again.

lmarzen, 2025-01-05:
I am talking to another user who is experiencing the same issue. Investigating..

lmarzen, 2025-01-05:
Try doing a clean build. Near the upload button first click 'Platformio: Clean', then upload again.
Let me know if that fixes it.

zombor, 2025-01-06:
Unfortunately not. I tried completely re-installing VS Code too just to make sure there was nothing laying around.

lmarzen, 2025-01-06:
The other user who recently reported the same issue, tried swapping their board out for another and made no other changes, and it resolved their issue.

I suspect it is a corrupted bootloader.

Someone had a similar issue and they claimed reflashing the bootloader fixed the issue.
https://github.com/sle118/squeezelite-esp32/discussions/271#discussioncomment-6481776
They link to this forum post with instructions:
https://forums.slimdevices.com/forum/user-forums/3rd-party-hardware/109412-announce-squeezelite-esp32-dedicated-thread/page119#post1627715

closed 2025-01-06 by lmarzen