lmarzen/esp32-weather-epd #143: N16R2

kingfisher1234, 2025-01-05:
I've got 2 versions of the FireBeetle 2 ESP32-E IoT Microcontroller

https://thepihut.com/products/firebeetle-esp32-e-iot-microcontroller-supports-wi-fi-bluetooth

and

https://thepihut.com/products/firebeetle-2-esp32-e-onboard-antenna?variant=43880039678147

The second (Purchaced a week after the first) is "N16R2"
This doesnt work with the BME280 sensor (I've just checked all the wiring and the only differnce is this version

lmarzen, 2025-01-06:
Hmmm, interesting. I'm not familiar with what makes the N16R2 different, I've never used it.

kingfisher1234, 2025-01-06:
I might have just  found the issue 
On the new "N16R2" board the pin that was #16 is now labled "NC"
I'll move the wires/update the config and see if that works  

https://wiki.dfrobot.com/_SKU_DFR1139_FireBeetle_2_ESP32_E_N16R2_IoT_Microcontroller#target_4

kingfisher1234, 2025-01-06:
I've just moved SCL to pin 12 and updated the config and it now works perfectly!

closed 2025-01-06 by lmarzen