lmarzen/esp32-weather-epd #145: Compatible with China region, supports Chinese character set, uses Amap Weather API

FengLiqiang996, 2025-01-11:
### Issue Description:
First of all, I would like to express my sincere gratitude for your hard work. This is an excellent project. I have successfully deployed it following your documentation, but encountered a few issues. Currently, the project uses the OpenWeatherMap API, which may not fully support weather queries for China, especially in terms of handling the Chinese character set. To better accommodate users in China, I suggest replacing the weather API with the Amap (Gaode) Weather API and ensuring support for Chinese characters in the display.

### Suggested Changes:
1. Replace the OpenWeatherMap API with the Amap Weather API.
   - The Amap Weather API better supports weather queries for China and returns data in the Chinese character set.
   - The request URL for Amap Weather is as follows:
     ```
     http://restapi.amap.com/v3/weather/weatherInfo?city=CityID&key=Amap_API_KEY
     ```

2. Adjust the weather data parsing logic to correctly handle the data structure returned by the Amap Weather API.
   - For example, temperature, weather descriptions, humidity, etc., need to be correctly extracted from the JSON data returned by Amap.

3. Ensure that the project supports the Chinese character set and avoids any issues with garbled or incorrect displays.

### Reference:
- Amap Weather API Documentation: [https://lbs.amap.com/api/webservice/guide/api/weather](https://lbs.amap.com/api/webservice/guide/api/weather)

### Heartfelt Thanks:
Once again, thank you for your hard work. This is a very valuable project. I have successfully deployed and used it, but encountered a few issues along the way. I hope these suggested changes will help the project better serve users in China. If you need further information or assistance, I would be happy to provide support. Thank you for your time and patience!

lmarzen, 2025-01-11:
Great suggestions. I already have planned to support those features, but I am a single developer doing this as a hobby for free, so don't expect these features soon.

1. Character Support - utf8 unicode support is a big challenge on embedded microcontrollers like the esp32 since they don't have enough flash to prerender entire fonts at the sizes we require. Rendering fonts on the fly is slow. I have some ideas on how to tackle this challenge and will hopefully implement it in the next year.

2. Support for additional APIs - This has been requested many times and is also planned, but other features have higher priority. Maybe in the next 2 years. If anyone knows C/C++ and wants to implement this in a modular way, I am glad to accept pull requests.

Hope this answers your questions, and thanks for the kind words.

-Luke

HKGDot, 2025-01-12:
Sounds great. It's 8 degrees in HK and still posting 13-14 degrees in OW API.

whtech, 2025-05-23:
[this](https://github.com/lmarzen/esp32-weather-epd/issues/17#issuecomment-2722092822) 
per this way, i add Chinese locale for the epd, it works but has little offset issues. 

[image]