lmarzen/esp32-weather-epd #149: Poor/no display, busy timeout with GDEY075Z08 3-color display

lipusal, 2025-01-17:
Hello! First of all, thank you very much for putting this code together, as well as for your extensive help with similar open issues, it's clear you care about folks getting their stuff to work. I'm hoping you'll be able to help me out too; I've put together hardware and software as guided in the repository, but my display is either not displaying at all, or displaying very faintly and/or flickering. Similar issues have pointed out to configuration issues or power, but before getting new hardware components or re-soldering I'd like to make sure I've done everything else I can.  Here's my setup:

### Hardware
- Board: FireBeetle 2 ESP32-E
- Display: Good Display GDEY075Z08 (3-color)
- Adapter: DESPI-C02 (RESE switch set to 0.47)
- No battery connected yet
- No BME connected yet

<details>
  <summary>Photos</summary>
  
#### Dim display + border box
[image]

[image]
[image]
[image]

</details>

### Configuration
I forked this to set my own config, the differences are [here](https://github.com/lmarzen/esp32-weather-epd/compare/main...lipusal:esp32-weather-epd:main).

## Symptoms
- Display will not turn on at all. Console shows `Busy timeout`
- Display shows a thick black border around
- Switching the DESPI switch to 3 and back to 0.47 caused one initial render once, but display remained faint after

### Console output
<details>
  <summary>Click to expand</summary>

```
Hard resetting via RTS pin...
--- Terminal on COM4 | 115200 8-N-1
--- Available filters and text transformations: colorize, debug, default, direct, esp32_exception_decoder, hexlify, log2file, nocontrol, printable, send_on_enter, time
--- More details at https://bit.ly/pio-monitor-filters
--- Quit: Ctrl+C | Menu: Ctrl+T | Help: Ctrl+T followed by Ctrl+H
ets Jul 29 2019 12:21:46

rst:0x1 (POWERON_RESET),boot:0x13 (SPI_FAST_FLASH_BOOT)
configsip: 0, SPIWP:0xee
clk_drv:0x00,q_drv:0x00,d_drv:0x00,cs0_drv:0x00,hd_drv:0x00,wp_drv:0x00
mode:DIO, clock div:2
load:0x3fff0030,len:1184
load:0x40078000,len:13232
load:0x40080400,len:3028
entry 0x400805e4
Connecting to 'xxxxxxxxxx'...
IP: 10.42.55.107
Waiting for SNTP synchronization..............................................
Thursday, January 16, 2025 21:30:51
Attempting HTTP request: api.openweathermap.org/data/3.0/onecall?lat=37.520&lon=-122.276&lang=en&units=standard&exclude=minutely&appid={API key}
  200 OK
Attempting HTTP request: api.openweathermap.org/data/2.5/air_pollution/history?lat=37.520&lon=-122.276&start=1737005455&end=1737091854&appid={API key}
  200 OK
Reading from BME280... BME not found
Busy Timeout!
_PowerOn : 20001265
Busy Timeout!
_Update_Full : 20001269
Busy Timeout!
_PowerOff : 20001107
Busy Timeout!
_PowerOff : 20001107
Awake for 92.077s
Entering deep sleep for 1667s
```
</details>

## Possible workaround
Connecting the display's 3.3v directly to the FireBeetle's makes the display vivid but flickery, similar to #123 (root cause there was an unsupported display; mine is supposedly supported). After a while, the display seems to stabilize. **Is this the expected behavior?** Please see [video](https://youtu.be/C3hYHiITJtE) (the display starts in its dim state and I press the reset button in the beginning)

<details>
  <summary>Console output during the video linked above</summary>

```
rst:0x1 (POWERON_RESET),boot:0x13 (SPI_FAST_FLASH_BOOT)
configsip: 0, SPIWP:0xee
clk_drv:0x00,q_drv:0x00,d_drv:0x00,cs0_drv:0x00,hd_drv:0x00,wp_drv:0x00
mode:DIO, clock div:2
load:0x3fff0030,len:1184
load:0x40078000,len:13232
load:0x40080400,len:3028
entry 0x400805e4
Connecting to 'xxxxxxxxxx'....
IP: 10.42.55.107
Waiting for SNTP synchronization..................
Thursday, January 16, 2025 22:06:25
Attempting HTTP request: api.openweathermap.org/data/3.0/onecall?lat=37.520&lon=-122.276&lang=en&units=standard&exclude=minutely&appid={API key}
  200 OK
Attempting HTTP request: api.openweathermap.org/data/2.5/air_pollution/history?lat=37.520&lon=-122.276&start=1737007589&end=1737093988&appid={API key}
  200 OK
Reading from BME280... BME not found
_PowerOn : 128993
Busy Timeout!
_Update_Full : 20001267
_PowerOff : 6051001
_PowerOff : 41001
Awake for 35.522s
Entering deep sleep for 1387s
```
</details>

----

Should I just leave the display connected directly to 3.3v? Anything else that I might've missed or misconfigured? Thanks in advance for your help 🙏

lmarzen, 2025-01-17:
Thanks for the kind words, but also for providing a detailed description of the troubleshooting steps you've taken.

The flickering is part of the expected refresh sequence of epaper displays. The more colors the epaper panel supports the more flickery the refresh. The black and white display refreshes in 4 seconds, whereas the 3 color you have is expected to take around 26 seconds. The 7-color panels take around 35s.

With regards to directly connecting to 3.3v, yes please do that. Others have experienced similar issues recently and this is the solution, I have been busy lately, but I have been meaning to update the wiring diagrams to indicate this. The original idea was to increase power savings by running the display off the power of a GPIO pin, which has enough power to be able to drive the black and white display, but as I have found out recently not the multi color displays. I will revert the wiring back to using 3.3v for powering the epaper in the coming days/week when I find time.

Let me know if you have any other questions.

lipusal, 2025-01-17:
Thanks for the quick and detailed response! That makes things easier then, will close this as resolved for now. I will keep an eye out for updates in case a solution is found to achieve the same power savings off of a GPIO port. In any case, thank you =)

closed 2025-01-17 by lipusal

itopaloglu83, 2025-01-31:
Very informative and well documented issue. Thank you.

For anyone checking this later, it looks like the [GxEPD2](https://github.com/ZinggJM/GxEPD2) library doesn't directly support the display part number GDEY075Z08. The supported panels list doesn't contain the part number and also the library owner made a comment (https://github.com/ZinggJM/GxEPD2/pull/61#issuecomment-1868896192) saying he doesn't have that display.

I wish I had known the amount of flicker 3 colored displays have. I would've bought the black-and-white one instead.

whtech, 2025-03-19:
> Very informative and well documented issue. Thank you.
> 
> For anyone checking this later, it looks like the [GxEPD2](https://github.com/ZinggJM/GxEPD2) library doesn't directly support the display part number GDEY075Z08. The supported panels list doesn't contain the part number and also the library owner made a comment ([ZinggJM/GxEPD2#61 (comment)](https://github.com/ZinggJM/GxEPD2/pull/61#issuecomment-1868896192)) saying he doesn't have that display.
> 
> I wish I had known the amount of flicker 3 colored displays have. I would've bought the black-and-white one instead.
> 
> Edit: The GDEY075Z08 display is not supported by the GxEPD2 library. https://forum.arduino.cc/t/waveshare-7-5-inch-3-colors-shows-nothing/1251108/2

GxEPD2 does support GDEY075Z08 3 colors. I tried it on FireBeetle 2 ESP32-E and ESP32 c3 dev boards. both of them worked.