lmarzen/esp32-weather-epd #156: Busy timeout

mattnueh, 2025-02-05:
Hi, I love the project but I'm having a little trouble. I got it to compile and upload through platformio on vscode. I'm not using the battery at this moment. I get no screen output, I'm using the despi board. I opened the serial monitor and discovered the bme280 wasn't detected, after the third bme280 replacement it reported as a success but then says:
Busy timeout! Power on, busy timeout, update full, busy timeout, power off, awake for 38.831s, entering deep sleep. I appreciate any help you can give me

lmarzen, 2025-02-06:
Sounds like there is an issue with your wiring. If share a photo I will take a look.

mattnueh, 2025-02-07:
[image]

[image]

I got it to work for just 1 refresh using this setup but that was testing it outside of the enclosure. I went to install it in the enclosure and it would no longer refresh and the contrast went from black to gray.

lmarzen, 2025-02-08:
Try connecting the 3.3v connection (currently D3) directly to the esp32's 3v3 pin. See if that fixes it, some users have reported that powering off he GPIO pin is insufficient. I have been meaning to update to documentation, but haven't gotten to it yet as I am in the middle of moving.

mattnueh, 2025-02-09:
I tried connecting the D3 pin to the 3V3 on the left side (usb pointing up)
and got the same result, then I tried connecting it to the 3v3 on the
right side and got the same error. Just to clarify, I ran the 3.3v wire
from the despi to the 3v3 pins (just to make sure I'm doing it right). I'm
going to try flipping the display cable again

mattnueh, 2025-02-09:
Whelp I've found a new issue, it seems my screen ribbon cable is torn
partially. :( i feel it's gonna be a few weeks before I get the new part

Elryth, 2025-03-05:
Thanks for your thread, it helped me troubleshoot my own issues (ribbon cable upside down). One thing I still don't have is a working BME, I have two, planning for a second build, neither are detected. I'm concerned I fried them soldering the headers, any idea why yours weren't working or anything to do/avoid when I order replacements?

mattnueh, 2025-03-05:
Hi, i appreciate your comment. I had the same issue with my bme280 not
working, luckily I had purchased a 3 pack, my hands were shaking too badly
due to the possibly early onset parkinsons which caused me to fry 2 of the
bme280 even when using a helping hands tool. I'm pretty sure the thick
solder i was using caused the issue as I had to hold it there for an extra
second or two which would cause extra heat to go in as well as melting more
solder which loved to go into the chip near the end pin. I also found
installing the pins upside down helped immensely. Like do the opposite of
the picture diagram to where the soldering iron is essentially hitting the
bare board if you mess up, just gives you a little more buffer. My 3rd one
worked after trying both tricks so I hope it'll work for you too!