lmarzen/esp32-weather-epd #158: Display not works

BDlawo, 2025-02-10:
Hallo,
I have an 800×480, 7.5inch E-Ink raw display with E-Paper_ESP32_Driver_Board (https://www.waveshare.com/wiki/E-Paper_ESP32_Driver_Board)

Problem: no display on the e-paper

Settings in the config.cpp:

// Pins for E-Paper Driver Board
const uint8_t PIN_EPD_BUSY = 25; // 5 for micro-usb firebeetle
const uint8_t PIN_EPD_CS   = 15;
const uint8_t PIN_EPD_RST  = 26;
const uint8_t PIN_EPD_DC   = 27;
const uint8_t PIN_EPD_SCK  = 13;
const uint8_t PIN_EPD_MISO = 12; // 19 Master-In Slave-Out not used, as no data from display
const uint8_t PIN_EPD_MOSI = 14;
const uint8_t PIN_EPD_PWR  = 26; // Irrelevant if directly connected to 3.3V

Display monitor (PlatformIO) after reset

rst:0x1 (POWERON_RESET),boot:0x13 (SPI_FAST_FLASH_BOOT)
configsip: 0, SPIWP:0xee
clk_drv:0x00,q_drv:0x00,d_drv:0x00,cs0_drv:0x00,hd_drv:0x00,wp_drv:0x00
mode:DIO, clock div:2
load:0x3fff0030,len:1184
load:0x40078000,len:13232
load:0x40080400,len:3028
entry 0x400805e4
Battery voltage: 284mv
Critically low battery voltage!
Hibernating without wake time!

Help, thanks

lmarzen, 2025-02-10:
Looks like you just need to disable battery monitoring in config.h.

BDlawo, 2025-02-10:
Thanks, now it works.

closed 2025-02-10 by lmarzen