lmarzen/esp32-weather-epd #159: How to proper prepare transtaltion locale file with diacretic signs?

androxkentaki, 2025-02-23:
Hello @lmarzen.

I assembled the weather station according to Your project - it is great, i love it.

I have prepare pl_PL locale translation file (polish).
But i have a problem with polish diacritics signs like ```ąęłńżźćóĄĘŁŃŻŹĆÓ```.

Can You explain hot to make them properly into .inc file?

According to used examples in project i have assumed, that i need go to Extended ASCII Table for Windows-1250
 https://www.ascii-code.com/CP1250

Find my character in table, check its ```OCT``` number and replace it in prepared ```locale_pl_PL.inc``` locale file  giving the prefix ```\``` at the beginning.
So e.g.

So for e.g. 
```Wschód słońca```

i have preprare

```Wsch\363d s\263o\361ca```

[image]
[image]
[image]

or ```Wilgotność```
as
```"Wilgotno\234\346"```

[image]
[image]

The translation is used properly, but diacritic signs is not displayed properly.
and on e-paper i have

[image]

I an using ```#define FONT_HEADER "fonts/FreeSans.h"``` in config.

Can You help me and explain what i have did wrong?

Is this a problem with my interpretation of usin ```OCT``` number ?
Ir maybe it is an issue with font (some letters are displayed properly?

I attach copy om by translation file (in txt).

[locale_pl_PL.txt](https://github.com/user-attachments/files/18932377/locale_pl_PL.txt)

When I finish working on it i want to prepare Pull Request with it to contribute this project.

AndrzejGerasimukARCHIMEDES, 2025-02-23:
I have discovered that IT seems like You have implement to the project Windows-1252.

Is there any way to add or change to Windows-1250?

androxkentaki, 2025-02-25:
Can anybody help me how to resolve this problem?

How to make this project to render on e-paper data with char set Central European Windows-1250 (CP1250) instead Western European Windows-1252 (CP1252)?

androxkentaki, 2025-02-26:
@lmarzen 
Can You confirm is there any way to implement this converter?
https://github.com/petrbrouzda/fontconvert8-iso8859-2/

It looks promising (but I am not a C programmer)

lmarzen, 2026-01-05:
It would be better to have a more complete solution for character sets. My goal is to eventually support the basic multilingual plane of utf-8.