lmarzen/esp32-weather-epd #160: [POSSIBLE SOLUTION] Low contrast on Waveshare 7.5" BW (FPC-C001)

AdamJSchofield, 2025-02-26:
When I initially set up this project on my 7.5" BW Waveshare EPD with the Waveshare ESP32 driver board (v3), the `DISP_BW_V2` macro called the `GxEPD2_750_T7` constructor. I got a really washed out image with low contrast, which is an issue I have seen on some other forums while looking around for a fix.

I noticed the startup sequence did not match the Waveshare demo that works just fine. Eventually I came across [this](https://github.com/ZinggJM/GxEPD2/blob/master/examples/GxEPD2_WS_ESP32_Driver/GxEPD2_WS_ESP32_Driver.ino#L102) in the GxEPD examples and noticed the FPC-C001 identifier. Lo and behold, my ribbon matches. So I swapped out the constructors and this project works like a charm.

I got my display from [Amazon](https://www.amazon.com/dp/B075R69T93). I am not sure if they ship different controller versions or what. I am more familiar with OLED display drivers. 

I would be happy to submit a PR with and additional macro for this display type, but wanted to ask first. Maybe one of them are being phased out and I just got unlucky?

jvherring, 2025-03-14:
I also have this display from amazon and would like to see exactly in detail how you got it to work correctly.

jvherring, 2025-03-16:
Did you also update renderer.h?

AdamJSchofield, 2025-03-16:
@jvherring Great, glad you got it working! Its good to know that it works with the DESPI as well. Just to confirm - you were having the low-contrast issue? Can you confirm that with the DESPI connected to 3.3v as you describe, and using the OLD display constructor, the issue persists? I'm just trying to make sure I understand whether its the 3.3v supply, or the startup instructions which vary by constructor. In my case I didn't have a problem with the power supply.

 It may be good to include some other defaults in this project for those of us with a slightly different display model. I'm working on some other projects at the moment, but when I come back to this one I'll submit a PR to make it more obvious for others who were in our position.