lmarzen/esp32-weather-epd #165: Connection Refused

kbloom, 2025-03-12:
My unit has been failing for several hours this morning with 

> One Call 3.0 API \
-1: Connection Refused

Is this a broader issue that others are seeing, or is it something specific to my network?

peterv99, 2025-03-12:
Nope, it seems global...
Cause is a certificate error:
```
[  2742][E][ssl_client.cpp:37] _handle_error(): [start_ssl_client():273]: (-9984) X509 - Certificate verification failed, e.g. CRL, CA or signature check failed
[  2757][E][WiFiClientSecure.cpp:144] connect(): start_ssl_client: -9984
  -1 Connection Refused
[  3157][E][ssl_client.cpp:37] _handle_error(): [start_ssl_client():273]: (-9984) X509 - Certificate verification failed, e.g. CRL, CA or signature check failed
[  3172][E][WiFiClientSecure.cpp:144] connect(): start_ssl_client: -9984
  -1 Connection Refused
[  3575][E][ssl_client.cpp:37] _handle_error(): [start_ssl_client():273]: (-9984) X509 - Certificate verification failed, e.g. CRL, CA or signature check failed
[  3590][E][WiFiClientSecure.cpp:144] connect(): start_ssl_client: -9984
  -1 Connection Refused
```

DingoSEAD, 2025-03-12:
Hello,
I have the same ! :(

lmarzen, 2025-03-12:
The certificate validation feature is on by default for enhanced security, but yes does require occasional manual update for now. If you wish to not have this trouble in the future you can disable it in config.h, uncomment `#define USE_HTTPS_NO_CERT_VERIF`.

https://github.com/lmarzen/esp32-weather-epd/blob/0dc305e6c4a5b1428c586d60ab24cb54f56aa2aa/platformio/include/config.h#L149-L151

I have pushed an updated certificate to the git repository.. [cert.h](https://github.com/lmarzen/esp32-weather-epd/blob/main/platformio/include/cert.h)

prisoner06, 2025-03-12:
Still not working with the new cert.h

Salothereal, 2025-03-12:
i have now downloaded and reconfigured the whole project and now it works again, thank you very much!

kbloom, 2025-03-13:
I chose to `USE_HTTPS_NO_CERT_VERIF `

jocoman74, 2025-03-14:
I copied the new Cert.h file, but I get this compile error?
src/main.cpp: In function 'void setup()':
src/main.cpp:253:20: error: 'cert_Sectigo_RSA_Domain_Validation_Secure_Server_CA' was not declared in this scope
   client.setCACert(cert_Sectigo_RSA_Domain_Validation_Secure_Server_CA);

minigl, 2025-03-19:
Hello,
I have the same problem as "@kbloom." I also downloaded the entire project again and recompiled it.
Of course, I made the necessary adjustments for myself. I didn't change anything in the code. Now I have the problem that the weather data isn't being retrieved (every 30 minutes). I don't get an error message on the display.
However, if I hold down the additional reset button for about 8-10 seconds, the weather data is retrieved and displayed correctly.
So, WiFi works. Did I forget a setting?
Does anyone have any ideas?

lmarzen, 2025-03-19:
@minigl Can you please copy and paste the terminal monitor output you get after you press the reset button? (the terminal output can be seen by pressing the terminal output button near the upload button in platformio/vscode). This should give us insight into how long the esp32 is deciding to sleep for. Also if you could share your sleep duration, and bed/wake time settings from config.cpp, that would be helpful in assisting troubleshooting your issue.

minigl, 2025-03-19:
Yes, of course:

--- Terminal on COM4 | 115200 8-N-1
--- Available filters and text transformations: colorize, debug, default, direct, esp32_exception_decoder, hexlify, log2file, nocontrol, printable, send_on_enter, time
--- More details at https://bit.ly/pio-monitor-filters
--- Quit: Ctrl+C | Menu: Ctrl+T | Help: Ctrl+T followed by Ctrl+H
ets Jul 29 2019 12:21:46

rst:0x1 (POWERON_RESET),boot:0x13 (SPI_FAST_FLASH_BOOT)
configsip: 0, SPIWP:0xee
clk_drv:0x00,q_drv:0x00,d_drv:0x00,cs0_drv:0x00,hd_drv:0x00,wp_drv:0x00
mode:DIO, clock div:2
load:0x3fff0030,len:1184
load:0x40078000,len:13232
load:0x40080400,len:3028
entry 0x400805e4
Battery voltage: 4112mv
Connecting to 'FRITZ!Box 6591 Cable AD'...
IP: 192.168.178.94
Waiting for SNTP synchronization.................................................
Wednesday, March 19, 2025 8:13:07 p.m
Attempting HTTP request: api.openweathermap.org/data/3.0/onecall?lat=51.438&lon=6.573&lang=de&units=standard&exclude=minutely&appid={API key}
 200 OK
Attempting HTTP request: api.openweathermap.org/data/2.5/air_pollution/history?lat=51.438&lon=6.573&start=1742325191&end=1742411590&appid={API key}
 200 OK
Reading from BME280...Success
_PowerOn: 186997

const int SLEEP_DURATION = 30;

const int BED_TIME  = 00; 
const int WAKE_TIME = 07;

lmarzen, 2025-03-19:
There should be a little bit more to the terminal output. after '_PowerOn: 186997'

minigl, 2025-03-19:
No, unfortunately not, that's all.

lmarzen, 2025-03-19:
Oh, I guess that's the issue then. And the display finishes updating? The log you shared cuts off as the display is powered on.

Have you double checked that the display and driver board selected in config.h match your hardware?

minigl, 2025-03-19:
Since I have a three-color display it is this setting:
(Display is initialized completely)

#define DISP_3C_B

#define DRIVER_WAVESHARE

// 3 COLOR E-INK ACCENT COLOR
// Defines the 3rd color to be used when a 3+ color display is selected.
#if defined(DISP_3C_B) || defined(DISP_7C_F)
 // #define ACCENT_COLOR GxEPD_BLACK
 #define ACCENT_COLOR GxEPD_RED
 // #define ACCENT_COLOR GxEPD_GREEN
 // #define ACCENT_COLOR GxEPD_BLUE
 // #define ACCENT_COLOR GxEPD_YELLOW
 // #define ACCENT_COLOR GxEPD_ORANGE

minigl, 2025-03-19:
Another message appeared:
"_PowerOn: 186997":

ets Jul 29 2019 12:21:46

rst:0x1 (POWERON_RESET),boot:0x13 (SPI_FAST_FLASH_BOOT)
configsip: 0, SPIWP:0xee
clk_drv:0x00,q_drv:0x00,d_drv:0x00,cs0_drv:0x00,hd_drv:0x00,wp_drv:0x00
mode:DIO, clock div:2
load:0x3fff0030,len:1184
load:0x40078000,len:13232
load:0x40080400,len:3028
entry 0x400805e4

lmarzen, 2025-03-19:
That's the start of the esp32 beginning execution again.

lmarzen, 2025-03-19:
At some point in the history of this project, I changed the default pinout. Maybe it's possinle your display is wired using a different pins than what are assigned in config.cpp.

minigl, 2025-03-19:
The display wiring matches the confi.cpp.

minigl, 2025-03-19:
I reloaded everything and made only the necessary adjustments. Here's the output on the monitor:

--- Terminal on COM4 | 115200 8-N-1
--- Available filters and text transformations: colorize, debug, default, direct, esp32_exception_decoder, hexlify, log2file, nocontrol, printable, send_on_enter, time
--- More details at https://bit.ly/pio-monitor-filters
--- Quit: Ctrl+C | Menu: Ctrl+T | Help: Ctrl+T followed by Ctrl+H
ets Jul 29 2019 12:21:46

rst:0x1 (POWERON_RESET),boot:0x13 (SPI_FAST_FLASH_BOOT)
configsip: 0, SPIWP:0xee
clk_drv:0x00,q_drv:0x00,d_drv:0x00,cs0_drv:0x00,hd_drv:0x00,wp_drv:0x00
mode:DIO, clock div:2
load:0x3fff0030,len:1184
load:0x40078000,len:13232
load:0x40080400,len:3028
entry 0x400805e4
Battery voltage: 4174mv
Connecting to 'FRITZ!Box 6591 Cable AD'...
IP: 192.168.XXX.XXX
Waiting for SNTP synchronization.......................
Wednesday, March 19, 2025 23:00:51
Attempting HTTP request: api.openweathermap.org/data/3.0/onecall?lat=51.43809&lon=6.57328&lang=en&units=standard&exclude=minutely&appid={API key}
  200 OK
Attempting HTTP request: api.openweathermap.org/data/2.5/air_pollution/history?lat=51.43809&lon=6.57328&start=1742335255&end=1742421654&appid={API key}
  200 OK
Reading from BME280... Success
_PowerOn : 185993
Busy Timeout!
_Update_Full : 20001262
_PowerOff : 232001
_PowerOff : 40001
Awake for 28.715s
Entering deep sleep for 1728s

lmarzen, 2025-03-19:
That looks promising. Looks like in should refresh in 30 minutes :). See if that works. If it does work let me know if you are able to determine what setting causes this bug.

minigl, 2025-03-19:
Unfortunately, there's no update after 30 minutes. What's going wrong here? I don't understand. When I first built the weather station, everything worked right away.

lmarzen, 2025-03-19:
I missed this...

Busy Timeout!
_Update_Full : 20001262
_PowerOff : 232001
_PowerOff : 40001

did your display update this time after pressing the reset button and waiting 20 seconds? The log seems to suggest the display isn't configured correctly.

minigl, 2025-03-19:
Yes, the display has been updated correctly. All data is displayed properly.

lmarzen, 2025-03-19:
If you change SLEEP_DURATION to 2 minutes. It may speed up our troubleshooting. We want to know whether the esp32 is waking up at all after the sleep period ends. To do this you need to have the serial monitor open, press the reset button, then wait 2ish minutes and see if it prints anything at all to the monitor.

minigl, 2025-03-19:
I get this message:

Reading from BME280...Success
_PowerOn: 185992
Busy timeout!
_Update_Full : 20001261
_PowerOff: 226000
_PowerOff: 40000
Awake for 27,898s
Entering deep sleep for 21686s

lmarzen, 2025-03-19:
You got that message after 2 minutes? So the display updated then? Looks like after that it is trying to sleep for 6 hours, maybe it is encountering it's bedtime (or maybe a bug). The timezone is used to help determine the bedtime. Im not sure what timezone your in, but let's try disabling the bedtime mechanic all together and see if that fixes it.

set BED_TIME and WAKE_TIME to 0.

minigl, 2025-03-19:
BED_TIME and WAKE_TIME are set to 0. Still no weather data updates. If I press the reset button for about 5-10 seconds, the display updates correctly to the current weather conditions. Then it stays dead.

minigl, 2025-03-19:
First of all, thank you very much for your patience. Since I'm writing from Germany, I'd like to ask if we can continue tomorrow? It's 0:37 CET here right now.

minigl, 2025-03-20:
Hello Imarzen,
are you here yet?

lmarzen, 2025-03-20:
You can either use git to checkout an older commit using the commit hash, or you can view the commits, select browse files and then I think you can download a .zip that way.

minigl, 2025-03-20:
I found your project from April 22, 2024, on my hard drive. It again links directly to the error:
One Call 3.0 API
-1: Connection Refused

I'm starting to suspect a hardware defect. I'll have to order a new FireBeetle and get back to you.

minigl, 2025-03-20:
This is exactly what my config.h looks like.

Still no proper functionality.

lmarzen, 2025-03-20:
I have tested it on my hardware and it's working (firebeetle esp32-e and despi-c02). If you share your configuration files config.h and config.cpp (BE SURE TO REDACT SENSITIVE INFORMATION), I will try running them on my hardware.

lmarzen, 2025-03-20:
That's fine

minigl, 2025-03-20:
Hello,
 
here is the configs...
 
 regards Michael
 
 

Gesendet: Donnerstag, 20. März 2025 um 19:07

minigl, 2025-03-20:
I have sent on email.

lmarzen, 2025-03-20:
You can't send attachments by responding to the github issue email chain. You can find my email on the website under my profile.

minigl, 2025-03-20:
Oh wow, i cant find your email. im to silly. where is it exactly?

lmarzen, 2025-03-20:
here: luke.marzen.me/about/
Dont want to post my email on github since it will be scraped, but it is on my website

minigl, 2025-03-20:
I have no problem, here is my email: minigl@web.de
please tell me your email and than i send you the config.c and config cpp

minigl, 2025-03-21:
I've sent both files again via email.

Just to be safe ;)

jocoman74, 2025-07-18:
> Happening to me too!
> I've just started seeing the following error:
> Air Pollution API
> Is this a cert issue like earlier? I've been using the
> USE_HTTPS_NO_CERT_VERIF workaround which doesn't appear to address this

Mine is back to normal!