lmarzen/esp32-weather-epd #171: I can reset the board and it works fine, but never gets past the first deep sleep.

foggyspruce, 2025-03-28:
Love the project! It is amazing and I am looking forward to getting my own setup perfectly. 

I can reset the board and it works fine, but never gets past the first deep sleep. Terminal serial output after a few minutes:
rst:0x7 (TG0WDT_SYS_RESET),boot:0x13 (SPI_FAST_FLASH_BOOT)
invalid header: 0xffffffff
invalid header: 0xffffffff
invalid header: 0xffffffff
invalid header: 0xffffffff
invalid header: 0xffffffff
invalid header: 0xffffffff
invalid header: 0xffffffff
invalid header: 0xffffffff
invalid header: 0xffffffff
invalid header: 0xffffffff
invalid header: 0xffffffff
invalid header: 0xffffffff
invalid header: 0xffffffff
invalid header: 0xffffffff
ets Jul 29 2019 12:21:46

I tried doing a full erase on the esp32 and reuploading. Still stuck here, but curious if I anyone else has run into this.

lmarzen, 2025-03-28:
I've seen this before. It's likely a corrupted boot load.

https://github.com/lmarzen/esp32-weather-epd/issues/142#issuecomment-2573832431

foggyspruce, 2025-04-20:
So I wasn't able to find the firmware necessary to reflash the bootloader from that previous thread (dead links and became time consuming), but I ended up buying a new board and everything works great now!

closed 2025-04-20 by foggyspruce