lmarzen/esp32-weather-epd #172: Lose wifi connection every 24 hours and won't reconnect

grooby, 2025-03-29:
Hi all,

I have a weird issue but not sure how to debug.  My weather station will disconnect from wifi and show "Wifi Connection failed" at around 24 hour mark.  The only way to recover is to either reset (short RST and GND) or re-flash the ESP32 ROM.  

Is there any special requirement for my wifi AP that I need to be aware of?

lmarzen, 2025-03-29:
Can you copy and paste the terminal output here? (open the terminal monitor by clicking the monitor button near the upload button in vscode/platformio).

grooby, 2025-03-31:
I am just seeing this.

--- Terminal on /dev/ttyUSB0 | 115200 8-N-1
--- Available filters and text transformations: colorize, debug, default, direct, esp32_exception_decoder, hexlify, log2file, nocontrol, printable, send_on_enter, time
--- More details at https://bit.ly/pio-monitor-filters
--- Quit: Ctrl+C | Menu: Ctrl+T | Help: Ctrl+T followed by Ctrl+H

lmarzen, 2025-03-31:
there should be more, try pressing the reset button so that the screen refreshes.

grooby, 2025-03-31:
Been messing around some more and noticed that if I reduce the wifi timeout to 1000ms, i get the wifi error screen almost immediately away.  I decided to bump up the timeout to 15000ms to see if this reduce this issue.

my hunch is that the weather station is placed right between 2 of my mesh AP node and most likely having trouble figuring out which one to talk to.

Additionally, I've grab the latest cert.h with the new cert and re-enable https_cert_verif.  I hope maybe this might help with things.

Will report back in another 24 hours.

grooby, 2025-04-01:
update:

encountered  "Wifi Connection Failed" after about 24 hours.  Plug it back into the computer and enable serial terminal to wait for the next refresh.  Refresh happened successfully but no output on the terminal.

grooby, 2025-04-01:
Closing this issue for now.  I think this has nothing to do with the code but more of ESP32 wifi design and how my home wifi is setup.

closed 2025-04-01 by grooby