lmarzen/esp32-weather-epd #175: Display refresh issue with DESPI-C02 and GDEY075T7

Aryeh95, 2025-04-11:
I recently started having the an issue where the display is really faded.
What's strange, is that during the refresh cycle, the contrast is normal, but when it's done refreshing, it's very faded. Do you know what might be the issue here?
(It was working for a long time before this issue suddenly started)
Video of the issue https://youtu.be/c86lxspP5bM

lmarzen, 2025-04-11:
Try switching the 3.3v pin from the despi that connects to a gpio pin on the esp32 to connect to the 3.3v pin on the esp32. See if that fixes it.

Aryeh95, 2025-04-11:
I just confirming your theory. I changed the hourly precipitation units from pop to inches and it resolved the issue.

[image]

blacktip2000, 2025-07-18:
[Cause]
When precipitation (bar chart) fills almost the entire graph area, ghosting occurs due to the characteristics of EInk displays.

[Fix Option 1]: Most reliable, but the bar chart becomes visually lighter.
https://github.com/lmarzen/esp32-weather-epd/blob/3cbd5a04bd8be6d668ade4a7fd84174946813a29/platformio/src/renderer.cpp#L1223
```
    for (int y = y1_t - 1; y > y0_t; y -= 2) 
    {
      for (int x = x0_t + (x0_t % 2); x < x1_t; x += 2)
      {
         if (((x / 2 + y / 2) % 2) == 0)
         {
          display.drawPixel(x, y, GxEPD_BLACK);
         }
      }
    }
```
:: before
[image]
:: after
[image]

(Or change the step values of -= 2 and += 2)

[Fix Option 2]: Given that extreme precipitation is relatively rare,
Change the UNITS_HOURLY_PRECIP_** option to a non-POP unit.

lmarzen, 2026-01-05:
Seems to be a duplicate of https://github.com/lmarzen/esp32-weather-epd/issues/62