lmarzen/esp32-weather-epd #176: Display is not turning on.

Woonpo, 2025-04-16:
Hi, I bought a screen from this link (recommended) https://www.aliexpress.com/item/3256802683908868.html
now they ship the 4 color version, in the description they still have the number GDEY075T7,
but the image and size option they describe as a 4-color version.

The version I received is **GDEM075F52**, is it possible to use this version or do I have to look for supported ones.

lmarzen, 2025-04-17:
Oh, they changed the product at the link. That's not very nice of them.

lmarzen, 2025-04-17:
When I go there I see that they have it listed as the 4 color display, but when you scroll down to the specifications it still says the model number for the intended product "Model Number GDEY075T7". You could try requesting a refund from the seller since the product specifications didn't match what was received.

[image]

lmarzen, 2025-04-17:
Unfortunately, the GDEM075F52 isn't on the list of supported displays for the display driver this project depends on, [GxEPD2](https://github.com/ZinggJM/GxEPD2).

lmarzen, 2025-04-17:
I updated the readme for now so others don't accidentally purchase this.

closed 2026-01-05 by lmarzen