lmarzen/esp32-weather-epd #178: Timeline under graph not displaying correctly

Artexman, 2025-05-05:
This is my first e-ink project and I ran into a couple issues, like the polarity on the battery connector that drove me nuts before coming across a comment on that issue....but now it seems the 24 hour timeline under the graph isn't displaying correctly. I never modified the code because 24 hour timeline was fine with me... Not a major deal just thought I would reach out and see if it was an easy fix... something I can't see being new. I'll try to upload a picture.

[image]

lmarzen, 2025-05-05:
Must be a bug, I'll investigate.

Artexman, 2025-05-06:
Thank you for looking into it.... I'm working 12 hours today and tomorrow... Will hook everything up and have a look at it first thing Wednesday morning. I bet I did try to change it to 12hr instead of leaving it on 24hr thinking it was going to be military time instead of 3pm to 3pm...so I'm almost positive I messed it up.... Thank you again... I will update you Wednesday with my value so you can correct me...🤦

lmarzen, 2025-05-06:
12 hour should be fine. Can you also tell me which locale you are using?

Artexman, 2025-05-06:
Yes... I'm using en_us

Artexman, 2025-05-08:
const char *HOUR_FORMAT = "%1%P";      // 24-hour ex: 01   23

Clearly it was my fault.... thank you again for your time and pointing me in the right directions... Awesome project!!

const char *HOUR_FORMAT = "%l%P";      // 24-hour ex: 01   23

Working fine now.... 

[image]