lmarzen/esp32-weather-epd #179: Add real-time GPS coordinates

sashoninov, 2025-05-11:
Hello, I want to install the station in my camper and it would be very useful if the coordinates for the forecast were taken from a GPS receiver, this way I would have an up-to-date forecast according to the location of the camper.

lmarzen, 2025-05-11:
That sounds like a great idea. GPS modules for the esp32 cost ~$15.

Here is a simple tutorial I found demoing how to use one.
https://randomnerdtutorials.com/esp32-neo-6m-gps-module-arduino/#neo-6m-gps-wiring-esp32

sashoninov, 2025-05-11:
Hello, I have a module to read data from it, I only need its TX connected to the RX of the ESP32, but since I am not a programmer, I need someone to write the code to make it work on this project.

lmarzen, 2025-05-11:
It might be easier to use your public IP to get an approximate location. Which wouldn't require any extra hardware. Do you use a cellular hotspot or do you use starlink or something else for your internet connection for your camper?

sashoninov, 2025-05-11:
i

[image]

 use the cellular network for the internet. In general, I have built a complete project in Esphome based on your project and further developed by me according to my needs for the camper, but in Esphome the project often crashes and I want to migrate from it. If you are interested, I can send it to you to look at.

sashoninov, 2025-05-12:
I really liked the idea of ​​saving a GPS receiver. The location can be extracted from here, for example http://ip-api.com/json/, but I still need help with the code. Thinking about it, it's not a bad option and for your project, it will eliminate the need to enter coordinates and the name of the location.

fbeels, 2025-10-28:
Here's something to get you started...

```
#include <WiFi.h>
#include <HTTPClient.h>
#include <ArduinoJson.h>

const char* ssid = "YOUR_SSID";
const char* password = "YOUR_PASSWORD";

void setup() {
  Serial.begin(115200);
  WiFi.begin(ssid, password);

  Serial.print("Connecting to WiFi");
  while (WiFi.status() != WL_CONNECTED) {
    delay(500);
    Serial.print(".");
  }
  Serial.println("\nConnected!");

  String ip = WiFi.localIP().toString();
  String url = "http://ip-api.com/json/" + ip;

  HTTPClient http;
  http.begin(url);
  int httpCode = http.GET();

  if (httpCode > 0) {
    String payload = http.getString();
    Serial.println("Response:");
    Serial.println(payload);

    // Allocate a JSON document
    const size_t capacity = JSON_OBJECT_SIZE(15) + 500;
    DynamicJsonDocument doc(capacity);

    // Parse JSON
    DeserializationError error = deserializeJson(doc, payload);
    if (error) {
      Serial.print("JSON parsing failed: ");
      Serial.println(error.c_str());
      return;
    }

    // Extract values
    const char* country = doc["country"];
    const char* regionName = doc["regionName"];
    const char* city = doc["city"];
    float lat = doc["lat"];
    float lon = doc["lon"];

    Serial.println("Parsed Location Info:");
    Serial.printf("Country: %s\nRegion: %s\nCity: %s\nLat: %.4f\nLon: %.4f\n", country, regionName, city, lat, lon);
  } else {
    Serial.printf("HTTP GET failed, error: %s\n", http.errorToString(httpCode).c_str());
  }

  http.end();
}

void loop() {
  // Nothing here
}
```

closed 2026-01-05 by lmarzen