lmarzen/esp32-weather-epd #180: Support for NOAA weather API (or other sources)?

pseddonxvi, 2025-05-31:
I've never been pleased with the accuracy of OpenWeather for where I live. I'm wondering how easy it would be to plug in a different API and use that instead.

kbloom, 2025-07-21:
While browsing the network graph of this repository,  I found [this fork](https://github.com/711brown/esp32-weather-epd/tree/Tempest) which may be a good example of adapting this device to a different API.

kbloom, 2025-07-21:
Here's another pull request that uses OpenMeteo instead of OpenWeatherMap:
https://github.com/lmarzen/esp32-weather-epd/pull/146
and a pull request that improves that fork further https://github.com/j-era/esp32-weather-epd/pull/1

So I guess this sort of thing is doable, and there is an abstraction separating the JSON response from the display code.

711brown, 2025-07-24:
> While browsing the network graph of this repository, I found [this fork](https://github.com/711brown/esp32-weather-epd/tree/Tempest) which may be a good example of adapting this device to a different API.

Thanks -- still a work in progress on fully adapting to the Tempest; but the NWS WWAs are working out nicely for me.