lmarzen/esp32-weather-epd #185: One Call API errors

ADYoungberg, 2025-07-18:
Don't know if this is an issue with with the software, assume there is an issue with the weather API server. But is anyone else having this message along with the one for air quality saying the API is having connection issues? Multiple displays I have are doing this and a coworker is having the same messages on his. On separate accounts and wifi networks. Sometimes it will display the weather information. Thank you!

[image]

lmarzen, 2025-07-18:
I'm observing the same issue today. I just tried making an api call from my smartphone, and it took almost 20 seconds to get a response. My display seems to be working again now. Seems to have been an issue on the side of OpenWeatherMap's servers.

If this issue persists or becomes more frequent, it should be mitigated by increasing the `HTTP_CLIENT_TCP_TIMEOUT` in config.cpp. The default is 10 seconds.

lmarzen, 2025-07-18:
As mentioned by @ADYoungberg, earlier today, the similar issue was reported https://github.com/lmarzen/esp32-weather-epd/issues/185 and was seemingly resolved without action needed from the users.

kbloom, 2025-07-21:
I continue to see this issue over the last couple days, so I'm going to increase `HTTP_CLIENT_TCP_TIMEOUT` in my configuration.

chozee99, 2025-09-27:
Has anyone tried to get an API key in the last few days?
I'm having some trouble.

lmarzen, 2025-09-27:
If you just created an API key, sometimes it can take a few hours up to 24 hours to activate.

chozee99, 2025-09-27:
Thanks, I'll try using it tomorrow.

Kr3tt3k, 2025-11-22:
Hello, is there any news on how to deal with this problem? I'm also getting this error message several times a day now. I've already increased the timeout, but I'm still getting the error regularly.