lmarzen/esp32-weather-epd #192: missing closing bracket in main.cpp

ASM-SW, 2025-09-05:
I downloaded the project today and brought it up in VSCode/Platform.io 
I did not change any of the #defines.  I think setup is missing a closing bracket.  Lines 304 and 5 are shown below:

```
  if(bme.begin(BME_ADDRESS, &I2C_bme))
  {
```
The bracket shown above is matching the line 360:
```
} // end setup
```

lmarzen, 2025-09-05:
Are you sure nothing got deleted? CI pipeline shows success on last build. https://github.com/lmarzen/esp32-weather-epd/actions/runs/16637840676

ASM-SW, 2025-09-05:
Please close.  It does compile.  The #if's have confused the plugin "Better C++ Syntax".  It is not ignoring code that should be ignored because of the #ifdefs.  I will have to write a problem report on the plugin.

Sorry for the request.  I went back and compiled it, and it compiles correctly.

closed 2025-09-05 by lmarzen